webcamoid/webcamoid #357: Webcamoid starts but makes nothing

tomasmueller, 2020-10-03:
Reporting a problem?

The application (appimage) starts without problem but makes nothing. Its not freezed but when i click to "preferences" or any other button nothing happens except the info button.

## Current Behavior

This is the first time i use the application.

## Expected Behavior

Do something.

## Your Environment

* Webcamoid information: 8.7.1 (appimage)
* Operating System information: Xubuntu 20.04 x64

# Questinons and other matters?

akvcam is installed

```
ls /lib/modules/$(uname -r)/extra/akvcam.ko*
/lib/modules/5.4.0-48-generic/extra/akvcam.ko
```

hipersayanX, 2020-10-03:
Try with the daily build, also just in case, press the **Play/Stop** button (8.7.1) or the **Play Sources** switch in the menu (daily).

olivpaulo, 2020-11-07:
The same happens to me, I only get a black screen. No play button, just a stop buton that does not stop anything. Software does not detect on board webcam in HP Notebook G62.

laikrai, 2020-12-11:
Same happened to me. I deleted the folder and then re-installed it and it worked.

Jalopy-Tech, 2021-01-28:
I had this problem. The issue is that in Linux, webcamoid will "play" its default camera device when opening. If the default camera device is not valid, it just sits there doing nothing.

To fix this problem, set the "Play webcam on start" setting to false as follows:

1. Edit the config file which should be in /home/[your user]/.config/Webcamoid/Webcamoid.conf
2. Change the line "playOnStart=true" to "playOnStart=false" (this line is in the "[StreamConfigs]" section. Save the changes.
3. Restart webcamoid.
4. Choose the correct camera device using the config sources button (second button from left).
5. Click the play button to start viewing the camera in webcamoid.

Once the correct camera device has been chosen, you can set the "Play webcam on start" setting back to true for automatic play if desired under Preferences -> General Options.

hipersayanX, 2022-02-15:
Problem solved in 9.0.0, closing.

closed 2022-02-15 by hipersayanX