webcamoid/webcamoid #359: New feature

reza-zoh, 2020-10-04:
Hey there, 
I just made a feature for my own issue which was my wide-angle webcam, with OpenCV, Numpy, etc. I came across your awesome program and I'm wondering if you can add this feature to your program?

As nowadays people are using digital tools to communicate, and are using cameras to make it more desirable to talk, I felt that such wide-angle cameras can cause problems to their privacy. I wrote a script that uses HAARCASCADE to center the person's upper body. Would you be interested to know more or it's just a boring idea? 
Wish you a great day, 
Reza

hipersayanX, 2020-10-05:
>I felt that such wide-angle cameras can cause problems to their privacy. I wrote a script that uses HAARCASCADE to center the person's upper body.

A [face tracking plugin](https://github.com/webcamoid/webcamoid/pull/329) was added and is available in the daily build, could it be that's the same feature you are asking for?

>Would you be interested to know more or it's just a boring idea?

I'm always open to new ideas of features :smile:

hipersayanX, 2022-02-15:
Feature added in 9.0.0, closing.

closed 2022-02-15 by hipersayanX