webcamoid/webcamoid #368: Trouble uninstalling

lipton1998, 2020-10-23:
Hello, I've tried Webcamoid but uninstalled it since I preferred the UI of a different Virtual Camera I installed. Then I used the other application but the "breathing" 'TRIAL' message from Webcamoid keeps overlaying my Virtual Camera output...
Very frustrating because Webcamoid keeps crashing after I reinstalled.  Anybody got an idea on how to fix this?

hipersayanX, 2020-10-24:
>Hello, 

Hi

>I've tried Webcamoid but uninstalled it since I preferred the UI of a different Virtual Camera I installed. Then I used the other application but the "breathing" 'TRIAL' message from Webcamoid keeps overlaying my Virtual Camera output...

Webcamoid is [Free Open Source Software](https://github.com/webcamoid/webcamoid/blob/master/COPYING), there is not [such](https://github.com/webcamoid/webcamoid/search?q=breathing&type=code) [messages](https://github.com/webcamoid/webcamoid/search?q=TRIAL&type=code) in Webcamoid, those messages are coming from the other app you have installed.

>Very frustrating because Webcamoid keeps crashing after I reinstalled. Anybody got an idea on how to fix this?

Try [resetting the settings](https://github.com/webcamoid/webcamoid/wiki/Reset-configs-and-starting-fresh).

sleepindawg, 2021-01-20:
I can't get webcamoid to uninstall from win10

hipersayanX, 2021-01-25:
@sleepindawg how did you installed? what was the error message.

sleepindawg, 2021-01-26:
No error message, It just doesn't uninstall. I ended up deleting the folder and running CCleaner to take out the registry entries.

hipersayanX, 2021-01-27:
@sleepindawg If there is not error message or any way to reproduce it, then that bug can't be solved.

hipersayanX, 2021-01-28:
@sleepindawg could it be that you have installed the daily build and when uninstalling you get a black cmd window for few seconds?

sleepindawg, 2021-01-28:
I suppose I could try reinstalling it and see if it will do the same thing again. I personally am only going to do this in the interest of assisting with development as people at gethub.com provide a great service to everyone with little compensation.

sleepindawg, 2021-01-28:
I doubt it was the daily build, but it could have been.

hipersayanX, 2021-02-06:
@sleepindawg will check it, I had a similar problem when testing the virtual camera, it could be a bug in QtIFW maybe.

closed 2021-02-16 by hipersayanX