webcamoid/webcamoid #370: Controlling PTZ camera

tskarthikeyann, 2020-11-06:
Is there a way to control the Pan Tilt Zoom of a PTZ camera which works with a standard UVC driver? http://www.imaginart.es/produccion_av/lumens/pdf/VC-B30U_DataSheet_English(LUI)-2017-1129.pdf is the camera I use. 

So far, I have written my own application in C# using directflow. Though I am able to control Pan, Tilt, Zoom, the camera doesn't react to the commands, sometimes.  

It would be great if you are able to help with this.

hipersayanX, 2020-11-08:
>It would be great if you are able to help with this.

I can't give you any help with that because I've not any camera like that, but in theory it should work.

tskarthikeyann, 2020-11-09:
Thanks for your response. After some serious debugging, found that the issue was because of USB 2.0. Everything starts working well as expected with USB 3.0. I can contribute to making the movement a little more smoother.. I will keep you posted on that. Thank you.

closed 2022-02-15 by hipersayanX