webcamoid/webcamoid #371: Unable to make the virtual camera work in Zoom (Ubuntu 20.04 LTS)

ytx21cn, 2020-11-07:
# Bug description

## Summary

I installed webcamoid and want to make a working virtual camera. I also installed the **v4l2loopback** virtual camera driver. Then I tried with a short video randomly pulled off from Instagram: [test video.zip](https://github.com/webcamoid/webcamoid/files/5504407/test.video.zip). However, when I opened Zoom to test it, it does not work at all.

## Current Behavior

The Zoom app does not recognize any virtual webcam at all.

## Expected Behavior

Zoom should show up the virtual web camera with the name and video feed set by myself.

## Steps to Reproduce

1. Install webcamoid from http://webcamoid.github.io/ (choose the **Installable, 64 bits** version, since I am using Ubuntu 20.04 LTS).
1. Install the v4l2loopback-dkms package: `$ sudo apt install v4l2loopback-dkms`.
1. Load the `v4l2loopback` module: `$ sudo modprobe v4l2loopback`.
1. Add a virtual webcam: open webcamoid, click on **Preferences** → **Virtual camera** → **Devices** → **Add**, and name it with a simple name "test-cam".
[image]
1. Switch to **Configure sources**, and add the short video I prepared previously. Then hit the **Play** button to start playing the video.
[image]
1. Open Zoom, and select **New meeting**. Then check the camera list. Here, only my built-in camera "HP TrueVision HD" is showing up, but no virtual camera is listed here.
[image]
1. I also installed `v4l2loopback-tuils` package. Here is the command line output of `v4l2-ctl --all`:
```
Driver Info:
	Driver name      : v4l2 loopback
	Card type        : test-cam
	Bus info         : platform:v4l2loopback-000
	Driver version   : 5.4.65
	Capabilities     : 0x85208003
		Video Capture
		Video Output
		Video Memory-to-Memory
		Read/Write
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps      : 0x85208000
		Video Memory-to-Memory
		Read/Write
		Streaming
		Extended Pix Format
		Device Capabilities
Priority: 0
Video input : 0 (loopback: ok)
Video output: 0 (loopback in)
Format Video Output:
	Width/Height      : 640/480
	Pixel Format      : 'YUYV' (YUYV 4:2:2)
	Field             : None
	Bytes per Line    : 1280
	Size Image        : 614400
	Colorspace        : sRGB
	Transfer Function : Default (maps to sRGB)
	YCbCr/HSV Encoding: Default (maps to ITU-R 601)
	Quantization      : Default (maps to Limited Range)
	Flags             : 
Streaming Parameters Video Capture:
	Frames per second: 30.000 (30/1)
	Read buffers     : 8
Streaming Parameters Video Output:
	Frames per second: 30.000 (30/1)
	Write buffers    : 8

User Controls

                    keep_format 0x0098f900 (bool)   : default=0 value=0
              sustain_framerate 0x0098f901 (bool)   : default=0 value=0
                        timeout 0x0098f902 (int)    : min=0 max=100000 step=1 default=0 value=0
               timeout_image_io 0x0098f903 (bool)   : default=0 value=0

```
1. Here is my output of `$ lsmod | grep v4l2`:
```
v4l2loopback           40960  0
videobuf2_v4l2         24576  1 uvcvideo
videobuf2_common       49152  2 videobuf2_v4l2,uvcvideo
videodev              225280  5 videobuf2_v4l2,v4l2loopback,uvcvideo,videobuf2_common
mc                     53248  4 videodev,videobuf2_v4l2,uvcvideo,videobuf2_common
```

It seems like that the virtual webcam has correctly loaded the virtual camera. But the softwares using webcam are not detecting the virtual webcam and receiving the inputs from it. Did I miss anything here?

## Your Environment

* Webcamoid information: 8.7.1, 64-bit
* Operating System information: Ubuntu 20.04 LTS, 64-bit

Leonpmlinux, 2020-11-07:
I have the same problem with v4l2loopback, it was updated on November 6 and everything works fine in OBS Studio, but when entering platforms like Zoom and Google Meet, it does not recognize the virtual camera.

hipersayanX, 2020-11-08:
v4l2loopback is very limited, many programs rejects the camera, you should try with [akvcam](https://github.com/webcamoid/akvcam) instead.

ytx21cn, 2020-11-22:
The v4l2loopback got a new update in the recent days.
```
$ apt list v4l2loopback-dkms -a
Listing... Done
v4l2loopback-dkms/focal-updates,focal-updates,now 0.12.3-1ubuntu0.2 all [installed]
v4l2loopback-dkms/focal,focal 0.12.3-1 all
```

My virtual webcam is working again!
[image]

closed 2020-11-22 by ytx21cn

hipersayanX, 2020-11-22:
@ytx21cn nice! :smiley: