webcamoid/webcamoid #373: How to "enable virtual webcam output?

gh-markt, 2020-11-09:
## Summary

I am experiencing what I believe may be a related issue to #273.   I had commented there, but as that issue was for the Mac, and my issue was with Linux, I was directed to create a new issue.  In a nutshell,  I am finding that other applications cannot get any video feed from a virtual camera crreated with webcamoid.

## Current Behavior
The virtual webcam is selectable by other apps as a video source, but only produces the following image in any application that tries to access the virtual camera:

[image]

I have not been able to find any way enable the webcam output in the webcamoid app to direct the source to the virtual webcam output, however.  

## Expected Behavior

I am not entirely sure if I am not looking in the correct place for how to enable webcam output, or if this is actually a bug

## Steps to Reproduce

I experience the same effects with zoom as well as in-browser apps which access the webcam whenI set the webcam device to use in that application to the virtual one, and do not experience any issues using the regular webcam while webcamoid is not running.

## Your Environment

 I am using linux 5.4.73, on a modern AMD processor, with akvcam as a virtual camera driver.

alexbossers, 2020-12-27:
I experience the same. Tried both akvcam as well as the loopback device.

Running current version of ubuntu 20.10 and kernel 5.4.0-58-generic on intel xeon.. 

- Using webcamoid repository version 8.6.1. Works by itself as a charm including live effects.
- Manually compiled akvcam 1.1.1 and installed using dkms. It is present next to the v4l2loopback device in lib/modules/5.4.0-58-generic/updates/misc/akvcam.ko

I configured webcamoid to output to a new virtual cam (see image).
[image]

My configuration is default and changing fake driver makes no difference.
[image]

Any logs I can provide to help troubleshoot this?

hipersayanX, 2020-12-28:
@alexbossers I need to solve some stability problems in akvcam, newer kernels added more fields to API structures, and v4l2-compliance did not give any clue of that, I will return with Webcamoid and this issue once I finish with akvcam.

leonardobsjr, 2021-06-29:
Any way to solve this? Ubuntu 24.04 Webcamoid 8.6.1 - Have this same screen on Teams :|

hipersayanX, 2022-02-15:
Problem solved in 9.0.0, closing.

closed 2022-02-15 by hipersayanX