webcamoid/webcamoid #374: Add output brings up a name textbox and format editing... but OK doesn't add any output

Xetaxheb, 2020-11-11:
## Summary

I've installed https://dl.bintray.com/webcamoid/webcamoid/windows/:webcamoid-daily-win64.exe on 11/11/2020 and everything else seems to work fine except "+ Add output" under video>outputs. Clicking "+ Add output" and then clicking "OK" makes no changes or additions. There are no new (virtual) webcam entries in any other program either.
There are 3 items in the Outputs list: "+ Add output", "X Remove all outputs", "[o] Set output picture"

## Current Behavior

Adding an output just does nothing... formats can be edited and added to the format list and removed fine, although they reset to default list on reopening the program. "Set output picture" seems to edit and save fine. Clicking on any of the format entries just opens to editing it, not 'selecting/choosing' it or anything similar. Clicking to edit and then save a format and then clicking OK also doesn't add an output successfully. 

## Expected Behavior

Add output > OK : new output generated?

## Your Environment

* Webcamoid information: webcamoid-daily-win64.exe on 11/11/2020
* Operating System information: Windows 10 Pro

# Questinons and other matters?
I'm also not seeing this UI on any screenshots I come across so I don't know if it's unique to windows or daily or what but this is what my program looks like (with a black image overlaid as a source)
[image]

Xetaxheb, 2020-11-11:
Here is what the various prompts I can access with "add output"> are, and none of them accomplish anything tangible or permanent.
[image]

hipersayanX, 2020-11-12:
You must install [akvirtualcamera](https://github.com/webcamoid/akvirtualcamera), otherwise it won't work. Also, both the daily build and akvirtualcamera are still in development and I did not tested it enough, I can't guarantee it will work.

ayunami2000, 2020-11-22:
I installed AkVirtualCamera and it still doesnt work

hipersayanX, 2020-11-22:
>Edit: I got a virtual device to show up with AkVcamManager.exe (quick note: you must AkVcamManager.exe update to actually save changes made with the tool)

Good! I must still document how to use the manager. Webcamoid of course does all the management from it's GUI.

>BUUUUT it doesnt show up in the application still

When you create a camera you must create it with:

```
AkVCamManager add-device "My virtual camera"
```
Then it will return an identifier string like ```AkVCamVideoDevice0```, then you must use that identifier to add at least one format to the camera:

```
AkVCamManager add-format AkVCamVideoDevice0 YUY2 640 480 30
AkVCamManager add-format AkVCamVideoDevice0 RBG24 640 480 30
```

Finally register the changes with:

```
AkVCamManager update
```

>(I'm guessing it isn't registered in webcamoid itself but is registered otherwise since I configured it via administrator command prompt instead of webcamoid, but this is just a guess and could be wrong)

The ```update``` sub-command must be run as administrator for registering the filter, the others can be run as a normal user because it stores the settings in the HKCU, but I'll change it to store the settings in the HKLM instead.

ayunami2000, 2020-11-22:
It's registered fully, but only shows up as an input video device in
webcamoid (& shows the placeholder image that I set it to show & uses the
formats, sizing, and frame rate that I set up) The whole output area just
doesn't seem to work. I also did figure out the running as administrator
part (& made a separate issue over on the AkVirtualCamera repository about
it) buuuut webcamoid just doesn't seem to want to work with the virtual
camera part.

hipersayanX, 2020-11-22:
@ayunami2000 That's probably because the service is not running, Webcamoid daily use the service to get the install path of the plugin. Because the service is not up, Webcamoid will detect the virtual camera as a normal camera.

setswservices, 2020-12-13:
I just installed the 64-bit version on a Windows 10 Pro system, build 19041.685, version 2004. Nothing happens when launched, though webcamoid shows up in the Task Manager, consuming 0% CPU time. I tried building it and launching and same thing.  Would you please tell me what version of webcamoid you have and what version of Windows?

hipersayanX, 2020-12-17:
@setswservices You are blocked for spamming the same question over and over in unrelated issues.

hipersayanX, 2022-02-15:
Problem solved in 9.0.0, closing.

closed 2022-02-15 by hipersayanX