webcamoid/webcamoid #375: How do I create and then enable the virtual webcam for zoom?

hlsimba, 2020-11-12:
Hey! I'm a bit of a noob and recently just switched from Windows to Linux (Ubuntu). I have downloaded webcamoid and the application works fine. I'd like to create a new virtual webcam so I can use it in zoom with all the cool effects and features. I tried reading the wiki guide but I can't understand one bit lol. Can someone help me with this?

hipersayanX, 2020-11-12:
You must install [akvcam](https://github.com/webcamoid/akvcam), it should work fine for Zoom, also I recommend you to install the master branch rater than the latest stable since I've patched some bugs, and I'm soon to make a new release.

hlsimba, 2020-11-12:
Awesome, Thank you for your great work!

hlsimba, 2020-11-12:
I tried installing the above and followed all instructions, however the app does not let me create the virtual webcam without freezing/crashing every time(see screenshot). Why is this happening?
[image]

hipersayanX, 2020-11-12:
I imagine that you have followed [the documentation](https://github.com/webcamoid/akvcam/wiki/Build-and-install#build-and-install) and have cloned the master, right? I haven't updated the documentation yet. The install step is now:

```
cd akvcam/src
make
sudo make install USE_DKMS=1
 ```
 
This way the driver will be managed by dkms by default.
Webcamoid 8.7.1 should be able to detect the driver without problem, if it still does not works, try with the [daily build](https://bintray.com/webcamoid/webcamoid/webcamoid/daily), I'm working on it right now and it should be stable (it should not take down the system, at least on the tests I made), even though there is a VIDIOC_REQBUFS EBUSY error when playing the vcam for second time, I need to solve this.

hlsimba, 2020-11-12:
I have followed "the documentation" and I have attempted all solutions. With the normal build, the application always crashes and with the daily build, it simply refuses to create a new virtual webcam (no response at all after I attempt to set it up). I know this may seem a little ironic asking you but, do you know any other software I can use to create a virtual webcam? I'm a simple man who just wants to use a filter (to blur our the rest of my room and only focus on my face) when my webcam is on in zoom lol.

hipersayanX, 2020-11-12:
>I have followed "the documentation" and I have attempted all solutions. With the normal build, the application always crashes and with the daily build, it simply refuses to create a new virtual webcam (no response at all after I attempt to set it up).

As I said before, there some bugs I need to solve yet, I can't give you an straight answer since I don't know what's the nature of the problem, because I need to investigate it.

>I know this may seem a little ironic asking you but, do you know any other software I can use to create a virtual webcam? I'm a simple man who just wants to use a filter (to blur our the rest of my room and only focus on my face) when my webcam is on in zoom lol.

https://alternativeto.net/

>Edit: I've also tried using obs as a virtual webcam using v4l2loopback and the obs plugin for it and there seems to be a problem. when I try the command ls/dev/video* it always fails to display a new video device. Seems to be an issue with Ubuntu 20.04 most probably.

The problem is v4l2loopback, I can't give much details since you don't seem to be a a technical user, but in short v4l2loopback support is [very limited](https://github.com/webcamoid/webcamoid/wiki/Virtual-camera-support#gnulinux) compared to akvcam.
But again, akvcam still needs some polishing to work well.

hlsimba, 2020-11-13:
Alright thanks!

hipersayanX, 2020-11-14:
I remembered

>With the normal build, the application always crashes

This one could be related to #331 which was fixed in the daily build.

Also I've fixed all known bugs with akvcam, it should be working fine now, I'll most probably add an installer and make a new release, and later I'll try to fix the related bugs in Webcamoid.

hlsimba, 2020-11-14:
Okay, let me try it out today. Thanks.

ytx21cn, 2020-11-22:
v4l2loopback-dkms got an update in the recent days.
My webcam is working again.

#371

[image]

closed 2021-02-16 by hipersayanX