webcamoid/webcamoid #379: Output resolution is always 640x480

ronisaacson, 2020-11-22:
Reopening #95. As of 8.5.0, the output resolution would always be 640x480 when recording mp4, but it followed the camera resolution if you used webm. Now with 8.7.1, I'm only getting 640x480 with both mp4 and webm output formats.

hipersayanX, 2020-11-22:
Webcamoid  will take the resolution of the source at the moment of start recording and will keep recording with that resolution even if you change the source, until you stop recording.
I think the solution would be adding an option for setting the recording resolution, either following source resolution or an user defined resolution.

ronisaacson, 2020-11-23:
Thank you but that's not the behavior I'm seeing, unless you mean the initial resolution when Webcamoid first opens? How do you change that?

Here's what happens: I open Webcamoid and it remembers that I last used my external camera (Logitech C930e). It always starts at YUYV 640x480 30fps. I change it immediately to MJPG 1280x720 30fps and the preview looks good. Then I click on the record tab and start recording.

With 8.5.0: if I choose an mp4 format, the output is always 640x480. If I use webm then it records at 1280x720 which matches the source resolution.

With 8.7.1: Using either mp4 or webm, it's always 640x480.

I can send you some clips to demonstrate this if you'd like. For both versions, I'm using Webcamoid on Centos 8 using your portable binary package.

hipersayanX, 2020-11-25:
>Here's what happens: I open Webcamoid and it remembers that I last used my external camera (Logitech C930e). It always starts at YUYV 640x480 30fps. I change it immediately to MJPG 1280x720 30fps and the preview looks good. Then I click on the record tab and start recording.

I'm not sure because it was a log time ago, but I've probably fixed that bug in the daily build.

>I can send you some clips to demonstrate this if you'd like. For both versions, I'm using Webcamoid on Centos 8 using your portable binary package.

Not necessary in this case, this bug does not depends on a specific file.

hipersayanX, 2022-02-15:
Problem solved in 9.0.0, closing.

closed 2022-02-15 by hipersayanX