webcamoid/webcamoid #381: Feature wish: Stream only a Window, or special region from the screen.

Dtrieb, 2020-11-26:
Stream only a Window, or special region from the screen.

ItsRogueRen, 2023-05-21:
Came to ask this same thing. I'm a Vtuber using Linux and virtual cams are the only way to share my 'webcam" with others, which is a tool that is severely lacking on Linux. Currently the best solution is to have 2 versions of OBS installed and use one for a virtual cam, which is a waste of resources. This app would be prefect if a specific window could be captured

hipersayanX, 2023-05-21:
@ItsRogueRen yeah I know, some of  the co-related features I want to add for 9.2 are window selection, mouse cursor capture, green screen, and if possible but not guaranteed, minimalist mode, and basic 2D avatar similar to the one used by pngtubers. Anyway, it will take some time because first at all I want to make some changes to the API.