webcamoid/webcamoid #382: virtual background in to do list?

ahahahaah, 2020-12-02:
hello 
i would like to ask if a virtual background option is or will be in the to do list of this software?
thanks

albinscode, 2020-12-12:
Hi @ahahahaah, please see #250 where the main discussion has started. 
I was interested too in this feature. 

Currently, I've used the last daily built of the 11 of December but got a runtime issue certainly due to dependencies not found on my system. I was able to build it on my own on Ubuntu 20.04 and get rid of some GUI components that were causing trouble. Looks like components related to new GUI currently developed.

By the way, the feature is still not implemented but an interesting one exists. It is to blur what's outside of a "face detection".

[image]

It can be done on face or other body parts.

[image]

Hope that helps!

hipersayanX, 2020-12-17:
Duplicate of #250, closing.

closed 2020-12-17 by hipersayanX