webcamoid/webcamoid #383: Problem with multiple instances of webcamoid

nejatafshar, 2020-12-13:
On Windows, when running multiple instances of Webcamoid to have multiple virtual webcams running concurrently, the output frame rates of webcams drops with increasing the number of instances. Does this problem have anything to do with AkVCamAssistant? Any hint how this could be solved?

setswservices, 2020-12-13:
Hi, nejatafshar.  I just installed the 64-bit version on a Windows 10 Pro system, build 19041.685, version 2004. Nothing happens when launched, though webcamoid shows up in the Task Manager, consuming 0% CPU time. I tried building it and launching and same thing.  Would you please tell me what version of webcamoid you have and what version of Windows?

nejatafshar, 2020-12-14:
@setswservices I use Windows 10 Enterprise 64 bit and compiled latest code of Webcamoid. Maybe you should delete `HKEY_CURRENT_USER\Software\Webcamoid` registry key and /share/config/Webcamoid/Webcamoid.ini in Webcamoid app dir.

hipersayanX, 2020-12-17:
It's normal, AkVCamAssistant work the same as a web server, when streaming, each client sends a notification to the assistant for informing that a new frame is available, the assistant redirects the message to each one of the target clients, the more streams you send, the more you increment the workload, the more you decrease the frame rate.
I doubt it can be solved, but if someone has an idea on how to solve it will be very welcome.

closed 2021-02-16 by hipersayanX