webcamoid/webcamoid #384: Getting webcamoid to work in Debian / KDE menus

atorrey, 2020-12-14:
I am running Debian (buster) w/ KDE.  The latest version in Synaptic is 8.5.0, which installs and starts fine.  It created an entry in the KDE applications menu, which worked fine.  I saw the note about the new version and downloaded the 'installable 64-bit version' 8.7.1.

I had some problems initially, but uninstalling all the versions and starting from scratch worked.

The install wanted to put the webcamoid directory in the root partition (BAD!) I switched it to /usr/share (which is where Synaptic put it)  I left the 'start webcamoid when install finishes button checked, and the program started, so I believe the install was correct.

However the 8.7.1 install program did NOT create an entry for the program in the KDE menu, so no obvious way to start it.

I attempted to manually create an entry using the KDE menu editor,  and have not been able to get it to start the program.  What do I need to put in the entry to make it work?

(also a minor bug, the install download also fails to bring down an Icon for the start menu - I finally grabbed the logo photo off the web page after a long time searching unsuccessfully in the installed files....)

Thanks,
ART

hipersayanX, 2021-10-21:
I was testing it and is working fine in the daily build/master branch, Webcamoid is installed in the home by default and appears in the Multimedia menu.

>also a minor bug, the install download also fails to bring down an Icon for the start menu - I finally grabbed the logo photo off the web page after a long time searching unsuccessfully in the installed files....

The icons are installed in _~/.local/share/icons/hicolor/_.

hipersayanX, 2022-02-15:
Problem solved in 9.0.0, closing.

closed 2022-02-15 by hipersayanX