webcamoid/webcamoid #387: Segfault if a UVC device is connected.

AndKe, 2020-12-19:
As subject says I have a UVC device (analog video capture device) that "just works" with any/all apps I tested, but displaying video from it crashed webcamoid instantly.

## Steps to Reproduce

If Webcamoid crashed, try to reproduce the crash several times to be sure where is the problem. Write the steps to reproduce the issue bellow:

1.  Open this
2.  Click that
3.  Drag those
4.  ...

## Suggestions and tips

How would you solve the problem?

## Your Environment
Ubuntu 20.10
device: Bus 005 Device 004: ID 534d:0021 MacroSilicon MS210x Video Grabber [EasierCAP]
dmesg:
[  3.479524] usb 5-2: new high-speed USB device number 4 using xhci_hcd
[  3.631340] usb 5-2: New USB device found, idVendor=534d, idProduct=0021, bcdDevice= 1.21
[  3.631343] usb 5-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  3.631345] usb 5-2: Product: AV TO USB2.0
[  3.631346] usb 5-2: Manufacturer: MACROSIL
[  3.631780] uvcvideo: Found UVC 1.00 device AV TO USB2.0 (534d:0021)
[  3.634216] uvcvideo: UVC non compliance - GET_DEF(PROBE) not supported. Enabling workaround.

The crash looks like:
(this is webcamoid starting up with this device attached)
```
./webcamoid.sh 
Qt: Session management error: None of the authentication protocols specified are supported
libv4l2: error getting pixformat: Invalid argument
qrc:/Webcamoid/share/qml/RecordConfig.qml:126:5: QML Label: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
Segmentation fault (core dumped)

 ```

FWIW: this video source works just fine with "cheese" , VLC, and any other app I tried, even M$ Teams

hipersayanX, 2020-12-28:
Give me the output of this command:

```
find /dev -type c -name 'video*' -exec echo Device {} \; -exec v4l2-ctl -d {} --list-formats-ext \;
```

AndKe, 2020-12-28:
I did not have v4l-utils installed, so I got errors
installed the v4l-utils and here is the output:

Device /dev/video2
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

Device /dev/video1
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'YUYV' (YUYV 4:2:2)
		Size: Discrete 720x480
			Interval: Stepwise 0.033s - 0.033s with step 0.000s (30.000-30.000 fps)
		Size: Discrete 720x576
			Interval: Stepwise 0.040s - 0.040s with step 0.000s (25.000-25.000 fps)
		Size: Discrete 640x480
			Interval: Stepwise 0.033s - 0.033s with step 0.000s (30.000-30.000 fps)
		Size: Discrete 320x240
			Interval: Stepwise 0.033s - 0.033s with step 0.000s (30.000-30.000 fps)
		Size: Discrete 160x120
			Interval: Stepwise 0.033s - 0.033s with step 0.000s (30.000-30.000 fps)
Device /dev/video0
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

hipersayanX, 2022-02-15:
Problem solved in 9.0.0, closing.

closed 2022-02-15 by hipersayanX