webcamoid/webcamoid #391: Visiting update config page resets update check interval

jaakristioja, 2021-01-06:
I was about to create a pull request, when I saw the note about CONTRIBUTORS.md in the pull request template. Since I really don't have time to deal with the legalese, here is a diff containing a description of the issue and a proposed fix,

```diff
Author: Jaak Ristioja <jaak@ristioja.ee>
Date:   Wed Jan 6 15:57:28 2021 +0200

    Fixed visiting the update config page resetting update check interval
    
    It seems that cbxCheckInterval.onCurrentIndexChanged() is also called after
    cbxCheckInterval.model is populated but before layout.onCompleted() is run.
    This effectively reset the check interval to "Daily" (and also ran the update
    check) when the combobox was constructed, overwriting the previous setting.

diff --git a/StandAlone/share/qml/UpdatesConfig.qml b/StandAlone/share/qml/UpdatesConfig.qml
index 2f39a62b..583844a9 100644
--- a/StandAlone/share/qml/UpdatesConfig.qml
+++ b/StandAlone/share/qml/UpdatesConfig.qml
@@ -58,6 +58,7 @@ Page {
                 }
 
                 cbxCheckInterval.currentIndex = ciIndex;
+                cbxCheckInterval.valueReadFromConfig = true;
             }
 
             Connections {
@@ -93,6 +94,7 @@ Page {
                 id: cbxCheckInterval
                 Layout.fillWidth: true
                 textRole: "description"
+                property var valueReadFromConfig: false
                 model: ListModel {
                     ListElement {
                         description: qsTr("Daily")
@@ -122,7 +124,7 @@ Page {
                 enabled: newVersion.checked
 
                 onCurrentIndexChanged:
-                    if (currentIndex >= 0)
+                    if (valueReadFromConfig && currentIndex >= 0)
                         updates.checkInterval = model.get(currentIndex).interval
             }
 
```

hipersayanX, 2021-01-08:
>I was about to create a pull request, when I saw the note about CONTRIBUTORS.md in the pull request template. Since I really don't have time to deal with the legalese

It basically says that the code you contribute to Webcamoid is still yours and you can reuse your code and contribute in other projects with whatever license you want, but you allow me to re-license the code your contributed to another [FOSS](https://en.wikipedia.org/wiki/Free_and_open-source_software) license (FOSS only), for example, if the code you contributed was under the terms of GPL I can re-license it under BSD if I ever need it, of course, preserving your authorship and you still can re-license it whatever you want.

jaakristioja, 2021-02-27:
I'm sorry, but I will not sign the Webcamoid Individual Contributor License Agreement, as it is simply not acceptable to me in its current form.

For what it's worth, I give Webcamoid permission to re-license my proposed changes as public domain.

hipersayanX, 2021-03-01:
>I give Webcamoid permission to re-license my proposed changes as public domain.

Closing then :man_shrugging: , I don't have any desire to release my own code as public domain.

closed 2021-03-01 by hipersayanX

jaakristioja, 2021-03-01:
@hipersayanX: That is not what I meant. But I guess I don't want to waste any more of my time trying to come up with new with legal tricks to help you integrate this trivial patch without risks.

Secondly, did you fix this bug in some other manner, or did we just accidentally discover a legal means to hinder you from fixing your code? 😆

hipersayanX, 2021-03-01:
>Secondly, did you fix this bug in some other manner, or did we just accidentally discover a legal means to hinder you from fixing your code? laughing

Nope, my code, my license, my rules. Your patches are based on my own code, and copyright laws takes into account the volume of incorporated work.
Also, as you said before:

>For what it's worth, I give Webcamoid permission to re-license **my proposed changes as public domain**.

Your changes are public domain :laughing: 

Also, the point of the CLA is to prevent a license lock like the [linux kernel](https://elinux.org/Legal_Issues#Kernel_is_licensed_GPL_v2_only) or [LibreCAD](https://github.com/LibreCAD/LibreCAD/issues/351).

For example, let suppose for a moment you decide to send a PR, your changes will be incorporated with the same license of the project (GPLv3+ as it is currently), if in a future I decide to re-license all or a part of the code to for example LGPL or MIT, or even public domain for what ever reason, without the CLA I'll need to ask you for an explicit permission for a re-license, if you don't give me the permission it will stay GPLv3+ forever. With the CLA, the contributor agree beforehand and I don't need to deal with all the bureaucracy later.

Now, the CLA I'm using doesn't steal your copyright, you still can reuse your code and contribute to others, with what ever license you want, from public domain to the most privative one. Again, the point of the CLA is to prevent a license lock on my own project.

Finally, the last thing I want to do is making all people agree with the CLA just for commenting, that does not looks good and sounds totally disrespectful.

jaakristioja, 2021-03-02:
Your code, your license, your rules... your potential loss, your unfixed bug. Your choice. I'm not complaining.

hipersayanX, 2021-03-02:
>I was about to create a pull request, when I saw the note about CONTRIBUTORS.md in the pull request template. Since I really don't have time to deal with the legalese, here is a diff containing a description of the issue and a proposed fix,

You were reluctant and did not wanted to collaborate from very beginning :man_shrugging: 
You need to change your attitude mate, just an advice, no hard feelings.

jaakristioja, 2021-03-04:
> You were reluctant and did not wanted to collaborate from very beginning man_shrugging
> You need to change your attitude mate, just an advice, no hard feelings.

@hipersayanX: These are baseless accusations. Just because I did not (and do not) want to enter into the legal agreement as written in CONTRIBUTORS.md does not imply I did not want to collaborate or was reluctant to collaborate.

I just find parts of the agreement questionable (2.4, 3(a), 4) and some of it unacceptable (6.1, 6.4 and 6.5; in part because I'm not proficient enough in (international) legal matters). This is why I filed a bug report instead of a pull request.

If I had not wanted to collaborate I would just have stopped at the pull request template and not filed this bug report. Neither would I have tried to find workarounds which I hoped would have allowed you to incorporate the proposed changes without me signing the agreement.

I believe I acted in good will, in good faith and tried to serve both of our interests.

I hope this helps you understand my position. No hard feelings.

hipersayanX, 2021-03-04:
>These are baseless accusations.

Again, based on this

>I was about to create a pull request, when I saw the note about CONTRIBUTORS.md in the pull request template. Since I really don't have time to deal with the legalese, here is a diff containing a description of the issue and a proposed fix,

and your hostile language, makes me think you did not wanted to collaborate from very beginning. 1 + 1 = 2

>I just find parts of the agreement questionable (2.4, 3(a), 4)

Why, explain.

>and some of it unacceptable (6.1, 6.4 and 6.5; in part because I'm not proficient enough in (international) legal matters).

Why, explain.

>EDIT: And as I stated in my first post, I originally simply did not have the time to go through the text of the agreement.

Let me guess, you click the accept button without ever reading the license :rofl:

jaakristioja, 2021-03-04:
> Again, based on this
>
>
> and your hostile language, makes me think you did not wanted to collaborate from very beginning. 1 + 1 = 2

In my first post just said that I found out about the CONTRIBUTORS.md file at the last moment and did not have the time to properly analyze it. But as I still wanted to help (as much as I could) at that moment, I thought it reasonable to file a bug report instead, especially as you could probably yourself come up with a better solution to the problem than what is contained in my trivial patch. It took me a considerable amount of time to study the code and find this one possible solution to this problem. I'm sorry that I didn't have the time to perhaps formulate a better bug report. I felt it would have been a waste to completely throw the patch I came up with. I've seen projects with CLAs that don't consider such trivial changes to require signing the CLA, so I don't believe it was unreasonable for me to include the diff in hopes that it would make things easier for you.

What exactly in my comments did you interpret as hostile? I currently fail to see any hostility from my part.

> Why, explain.

I decline, mostly because I perceive the nature of your previous two comments to be in part accusatory. I hope I'm mistaken or that you reconsider.

>> EDIT: And as I stated in my first post, I originally simply did not have the time to go through the text of the agreement.
>
> Let me guess, you click the accept button without ever reading the license rofl

I hope you did not mean to insult me by saying this. I also fail to see your statement as funny, especially considering that I did in fact not rush to accept your license agreement, but instead later took the time to analyze it more thoroughly and then decisively declined to enter into agreement.

In any case, please forgive me if I have misinterpreted you.

hipersayanX, 2021-03-09:
>I've seen projects with CLAs that don't consider such trivial changes to require signing the CLA, so I don't believe it was unreasonable for me to include the diff in hopes that it would make things easier for you.

Most small projects doesn't really care about copyright issues, they will accept any code from anyone just to get the things fixed and working, ignoring completely that their project can be copyright locked just for accepting a small piece of code (obviously depending on how much substantial was the contribution).
For example, let say you have licensed your program as GPLv2 only, now let say you want to add a new feature to your program, but that feature requires a very specialized, complex and hard to maintain code, then you will go looking for libraries that handle that case, deriving all the work to a third-party meaning less work and maintenance cost, now imagine for a moment that all library offerings are either poorly maintained BSD licensed code, privative and impossible to use libraries, or a library that perfectly fits the case but... is licensed as GPLv3.
At this point you can:

1. Write the library from scratch, which mean wasting a lot of development time.
2. Go and cope with some crapy library that barely handle the case.
3. Or just re-license your code to GPLv3 to comply with the library.

The third option the by far the best solution, but since you decided to accept code from everywhere now your own code is copyright locked and you will have to go with 1 or 2.
A CLA, or at least the one I'm using, was made to prevent this case allowing me to re-license the code and just that.
If you want some examples of projects using a CLA or equivalent:

**The Free Software Foundation and GNU project**

https://www.fsf.org/bulletin/2014/spring/copyright-assignment-at-the-fsf
https://www.gnu.org/licenses/why-assign.html

**The Linux kernel**

https://elinux.org/Developer_Certificate_Of_Origin

**KDE**

https://ev.kde.org/rules/fla/

**GNOME**

https://wiki.gnome.org/action/show/FoundationBoard/Resources/CopyrightAssignment
https://wiki.gnome.org/FoundationBoard/Resources/CopyrightAssignment/Guidelines

**Mozilla/Firefox**

https://www.mozilla.org/en-US/about/governance/policies/commit/requirements/

**Chromium**

https://www.chromium.org/developers/contributing-code/external-contributor-checklist

**Qt**

https://www.qt.io/legal-contribution-agreement-qt

**Ubuntu**

https://ubuntu.com/legal/contributors

>What exactly in my comments did you interpret as hostile? I currently fail to see any hostility from my part.

Knowing that you need to accept the CLA to contribute, even though you decided to try to bypass and disrespect my decision based on a weak argument:

>Since I really don't have time to deal with the legalese

And doing a big storm of nothing. Any of the other projects mentioned above will just have kicked or ignored you for doing the same you tried to do with my project.

>I decline, mostly because I perceive the nature of your previous two comments to be in part accusatory. 

Accusatory of what? I've explained my point on using a CLA over and over with arguments.

>I hope you did not mean to insult me by saying this.

Oh c'mon, don't be thin skin, is just a joke to relax the debate :smile: 

>In any case, please forgive me if I have misinterpreted you.

Ok