webcamoid/webcamoid #392: Webcamoid ruined my camera

Yatoya001, 2021-01-07:
I am actually typing this comment out on a separate computer to show the difference of what this software did. I would like a remedy because it also effected the built in camera on my Dell Latitude 7591 2in1.

[image]
2 Brand new cameras Aukey PC-LM1E the one on the right was what happened after installing Webcamoid.
I know this picture displays the 2 cameras in a different piece of software, don't criticize it.

I have applied the fix suggested in the Webcamoid destroyed my camera and microphone #294 .
I have also applied the Skype suggested fix, which does help. #148 

But it is apparent that Webcamoid did something to the camera itself.

Please provide a solution.

Yatoya001, 2021-01-08:
Please disregard the above, I had 2 plastic films on my camera lens I found out, I never noticed it on the one for the left but for the one on the right only 1 plastic film was removed. I am learning also that every camera software applies its own filters which effect the camera in there own ways, that is why the fix for #148 helped.

Yatoya001, 2021-01-08:
Just to post that its fixed.
[image]

On the same computer that used to have Webcamoid installed.

hipersayanX, 2021-01-08:
Your issue is the same as  #148.

>But it is apparent that Webcamoid did something to the camera itself.

Just in case, Webcamoid doesn't touch nothing related to the firmware or the hardware itself, Webcamoid just communicate with the hardware using the API provided by the operating system, so the hardware itself is not really affected or damaged.
What happened with that issue is that Webcamoid wrongly disabled the automatic controls of the camera (auto exposure, auto focus, etc), and that makes the picture look bad.
As said in that issue, one solution is setting all auto. controls on again in Skype or your camera software, and the other is just disconnecting and connecting the camera again (or turn off and on your laptop), at least in my experience the webcam does not save the settings internally.

Yatoya001, 2021-01-08:
Thank you for replying even after I made a closing comment.

closed 2021-01-08 by Yatoya001

hipersayanX, 2021-01-08:
Sure, no problem, I'm glad you have solved your problem at least :smile: