webcamoid/webcamoid #393: Provide signed releases for Windows installer

Abdull, 2021-01-11:
With all the hacks going on, users need trust in installing software. Currently, the webcamoid installer for Windows is not signed, holding me and probably others back in trying out webcamoid.

It would be great if the Windows installers were signed.

hipersayanX, 2021-01-11:
Ok, will you pay for it? because I won't.

cjbarth, 2021-01-11:
In case you're wondering, [here](https://stackoverflow.com/a/49696454/271351) is how to do it with a link to how much it costs [here](https://www.certum.eu/en/code-signing-certificates/) or [here](https://cheapsslsecurity.com/sslproducts/codesigningcertificate.html) @Abdull . There are links on the preferred way to donate on the main page of this project. Honestly, I'm interested in this, but only with the new v9 release which has a feature I'm really waiting for: Face Tracking.

closed 2022-02-15 by hipersayanX