webcamoid/webcamoid #394: Virtual camera is reason of crash all chromium-based browsers and suspending OS at all

contentadore, 2021-01-13:
# Reporting a problem?

## Summary

I created virtual camera then tried to delete them but had suspension of the program
and now it's reason of crash all chromium-based browsers and suspension OS at all because they are trying to connect to the camera.
How can I delete the virtual camera manually from terminal on Ubuntu 20.04?

## Current Behavior

OS Ubuntu gets suspend at all. Mouse isn't working. Need to reboot.

## Expected Behavior

Should work without suspention of programs.

## Steps to Reproduce

If Webcamoid crashed, try to reproduce the crash several times to be sure where is the problem. Write the steps to reproduce the issue bellow:

I'm not sure. I created virtual camera then it worked fine some period, but then got error with all programs that can connect with the virtual camera.

## Suggestions and tips

## Your Environment

* Webcamoid information: 8.7.1 x86_64
* Operating System information: Ubuntu 20.04 x86_64
* Any other useful information: 

# Want a new feature?

# Questinons and other matters?

How to delete webcamoid and all virtual cameras?

contentadore, 2021-01-13:
I deleted all files from ~/webcamoid/ via uninstaller, but virtual camera was left.

I found solution how to remove akvcam driver from Ubuntu https://unix.stackexchange.com/questions/162455/remove-a-manually-installed-driver-wireless-usb

```
$ lsmod | grep akvcam

$ sudo find /lib/modules/$(uname -r) -name akvcam.ko

$ sudo rm /lib/modules/x.x.x-xx-generic/updates/dkms/akvcam.ko
```

hope it helps

closed 2021-01-13 by contentadore

hipersayanX, 2021-01-25:
You don't explain what version and how did you installed akvcam, did you installed from Ubuntu repositories? did you used the installed? did you installed manually?

>I found solution how to remove akvcam driver from Ubuntu https://unix.stackexchange.com/questions/162455/remove-a-manually-installed-driver-wireless-usb

The next time you update your kernel, akvcam gonna be rebuild again because you did not uninstalled properly, don't follow a random hacky solution if you don't understand what you are doing. 
If you installed akvcam through Ubuntu repositories you must use apt to uninstall it.
If you installed akvcam through the installer you must use the installer located in /opt to uninstall it.
If you installed akvcam manually you must uninstall it [using DKMS](https://github.com/webcamoid/akvcam/wiki/Build-and-install#build-and-install).

>I deleted all files from ~/webcamoid/ via uninstaller, but virtual camera was left.

Again don't try hacky solutions. Webcamoid and akvcam are two independent programs, and can be used and installed/uninstalled independently, Webcamoid is not responsible in anyway for uninstalling akvcam.
What you erased was Webcamoid's settings that has nothing to do with akvcam.