webcamoid/webcamoid #395: Video recording has no sound

cmonty14, 2021-01-21:
Hello,

I have installed webcamoid 8.7.1-2 in Arch Linux; there were no issues during installation.

When I start webcamoid, my webcam is identified correctly and I can use the app for taking pictures and videos.

However, the video has no sound.

My webcam has a build-in stereo mic, and I can see that this is working in the audio settings.

Is there a setting in webcamoid that must be checked when taking a video with the webcam using the webcam's mic?

THX

leandrolnh, 2021-07-27:
I've seen this happening with FFmpeg version 4.4, no audio recorded. Version 4.3.2 does record sound.

papahippo, 2021-12-21:
I'm also unable to capture sound using webcamoid 8.8.0; unsure what libraries I am using.
This is under Mx Linux 4.3 KDE version (baed on Debian 10)..
How do suggest I diagnose this further?

papahippo, 2021-12-21:
... following up..
I have confirmed that that guvcview still works but of course it not as good on some points as webcmoid.
In particular, I like the facility to matrix-map the video so I can put my physical camera in 'portrait mode'.

hipersayanX, 2021-12-21:
@papahippo this bug should have been fixed already in the daily build.

papahippo, 2021-12-22:
Thanks. I'll try to take the daily build on board here.

papahippo, 2021-12-22:
Hi, I've installed the daily build but am encountering the following errors:
<
gill@luna:~/webcamoid$ sh webcamoid.sh 
webcamoid: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /home/gill/webcamoid/lib/x86_64-linux-gnu/libQt5Quick.so.5)
webcamoid: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /home/gill/webcamoid/lib/x86_64-linux-gnu/libQt5Widgets.so.5)
webcamoid: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /home/gill/webcamoid/lib/x86_64-linux-gnu/libQt5Gui.so.5)
webcamoid: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /home/gill/webcamoid/lib/x86_64-linux-gnu/libQt5Qml.so.5)
webcamoid: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /home/gill/webcamoid/lib/x86_64-linux-gnu/libQt5Core.so.5)
webcamoid: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /home/gill/webcamoid/lib/x86_64-linux-gnu/libpng16.so.16)
webcamoid: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /home/gill/webcamoid/lib/x86_64-linux-gnu/libicui18n.so.66)
webcamoid: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /home/gill/webcamoid/lib/x86_64-linux-gnu/libicuuc.so.66)
gill@luna:~/webcamoid$ 
>
This looks like something I should be able to install the missing library myself, but when I searched for how to do it, I found (patronizing or helpful?) messages suggesting I should not be wanting to to this.
What is your advice?

papahippo, 2021-12-22:
oh, it would appear that my attempt at formatting went wrong.  This may be more legible:

gill@luna:~/webcamoid$ sh webcamoid.sh 
webcamoid: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /home/gill/webcamoid/lib/x86_64-linux-gnu/libQt5Quick.so.5)
webcamoid: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /home/gill/webcamoid/lib/x86_64-linux-gnu/libQt5Widgets.so.5)
webcamoid: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /home/gill/webcamoid/lib/x86_64-linux-gnu/libQt5Gui.so.5)
webcamoid: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /home/gill/webcamoid/lib/x86_64-linux-gnu/libQt5Qml.so.5)
webcamoid: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /home/gill/webcamoid/lib/x86_64-linux-gnu/libQt5Core.so.5)
webcamoid: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /home/gill/webcamoid/lib/x86_64-linux-gnu/libpng16.so.16)
webcamoid: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /home/gill/webcamoid/lib/x86_64-linux-gnu/libicui18n.so.66)
webcamoid: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /home/gill/webcamoid/lib/x86_64-linux-gnu/libicuuc.so.66)
gill@luna:~/webcamoid$

papahippo, 2021-12-22:
Sorry, can't fix the layout. Can't remove the ugly mess above either

hipersayanX, 2021-12-22:
@papahippo which distro and version are you using.

papahippo, 2021-12-22:
See below - but quick answer is: MX Linux based on  Debain 10 (buster).
[code]
System:    Host: <filter> Kernel: 5.6.0-2-amd64 x86_64 bits: 64 compiler: gcc v: 8.3.0 
           parameters: BOOT_IMAGE=/boot/vmlinuz-5.6.0-2-amd64 root=UUID=<filter> ro quiet splash 
           init=/lib/systemd/systemd 
           Desktop: KDE Plasma 5.14.5 wm: kwin_x11 dm: SDDM Distro: MX-19.4_KDE_x64 patito feo August 16  2020 
           base: Debian GNU/Linux 10 (buster) 
Machine:   Type: Desktop Mobo: ASUSTeK model: P8H67-M PRO v: Rev 1.xx serial: <filter> BIOS: American Megatrends v: 0902 
           date: 03/04/2011 
[/code]

papahippo, 2021-12-22:
I think I need to download some debian 11 versions of libraries - not just glibc - to the special  library sub-directory of the downloaded webcamoid.... not sure which.

hipersayanX, 2021-12-22:
@papahippo yes, you will need to upgrade to Debian 11 at least, just downloading glibc won't work.

hipersayanX, 2022-02-15:
Problem solved in 9.0.0, closing.

closed 2022-02-15 by hipersayanX