webcamoid/webcamoid #397: Daily Build .AppImage Not opening

antintin0, 2021-01-23:
I've been trying to figure out a reliable and convenient way of blurring / replacing my background for zoom in kde neon on my crappy 6th gen i5 laptop. I've tried practically all the methods discussed in issue #250, but most lower my fps far too much, are too unintuitive to launch every single time I have a meeting, and crash a decent amount. 
I'm now trying to use the blur / chroma key feature on the daily build for linux, but, whether I use the .run or .appimage, webcamoid never actually starts. It does appear in my process monitor and takes up about 65 mb of ram, but the GUI just never appears. 
If anyone knows why, I would greatly appreciate it.

hipersayanX, 2021-01-25:
Try with [this](https://github.com/webcamoid/webcamoid/wiki/Reset-configs-and-starting-fresh).

hipersayanX, 2022-02-15:
Problem solved in 9.0.0, closing.

closed 2022-02-15 by hipersayanX