webcamoid/webcamoid #402: MP4 videos saved at text files

fido1982, 2021-02-04:
Good afternoon.

I am running Ubuntu and trying to get to grips with Webcamoid. 

One issue I am having relates to when I try and save a video (using my integrated webcam to film). I wish to save in MP4 format but it is saving as a text document instead.

For reference, it is not because I don't have the saved file assigned to a suitable application (MP4's are assigned to open in VLC player). 

I am really stumped so any feedback would be very well received.

hipersayanX, 2021-02-06:
I really don't understand this bug. What you are reporting does not seems possible.

You saved the file manually as *.txt file? or Webcamoid did it?
It was saved with a *.txt extension or what?
Did you have selected MP4 format in the left bar?
Does that file had binary data inside or its showing some error?
Which version of Webcamoid? 
Which version of Ubuntu?

hipersayanX, 2021-02-16:
No answer, not reproducible, closing.

closed 2021-02-16 by hipersayanX

murgascarrillo, 2023-01-03:
Hello!
I had a similar issue as fido1982 described.
Following your question I will answer in my specific situation:
* Webcamoid saved the video file as .mp4 but the OS is recoginizing it as it was a .txt. When opening it Celluloid pops up prompting: "unrecognized file format".
* I did selected MP4 format in te left bar.
* It's an empty file what is being created.
* Webcamoid 8.8.0.
* Linux Mint 21.

Thanks for making this up.

hipersayanX, 2023-01-04:
@murgascarrillo can you reproduce the same behavior with Webcamoid 9.