webcamoid/webcamoid #405: Webcamoid for Windows stopped working - OBS Studio Virtual Cam Use

WhiskeyMonkey, 2021-02-08:
Tried Webcamoid 32 bit and 64 bit - installed each in separate directories.  Both worked nicely in Windows 10, current edition.

Ran OBS Studio in virtual webcam mode.  Tried to select OBS studio virtual cam in Webcamoid 32 bit and Webcamoid crashed.  Windows reports "webcamoid has stopped working.".  

- Stopped OBS Studio and tried to restart both 32 then 64 bit - both Windows reported as not working.
- Rebooted ("Restart") Windows 10 - same results.  
- Uninstalled both versions of Webcamoid then reinstalled 64 bit- same results.
- Did a cold reboot (Close Down, wait, & reboot ) - same results.

Any suggestions?

WhiskeyMonkey

hipersayanX, 2021-02-08:
If Webcamoid does not starts try deleting _[HKEY_CURRENT_USER\Software\Webcamoid](https://github.com/webcamoid/webcamoid/wiki/Reset-configs-and-starting-fresh#windows)_.
I you got Webcamoid working, go to Preferences/Output/Remove All. If that does not works try following the steps in #178.

>Does not crash but does not have access to the video sources such as the integrated webcam.

The virtual camera should not affect other cameras but if it was the case, try [reinstalling the driver](https://support.microsoft.com/en-us/windows/update-drivers-in-windows-10-ec62f46c-ff14-c91d-eead-d7126dc1f7b6).

hipersayanX, 2022-02-15:
Problem solved in 9.0.0, closing.

closed 2022-02-15 by hipersayanX