webcamoid/webcamoid #407: The virtual camera doesn't work in Firefox/Edge

GaGag23, 2021-02-20:
In Firefox and Microsoft Edge the virtual camera isn't detected. Webcamoid works as I tested it in Discord. I tried resetting the configuration and reinstalling webcamoid. 
I think it's a Windows issue because in Ubuntu 20.10 it works fine.

* Webcamoid information: Webcamoid 8.8.0 (32 bit) 
* Operating System information: Windows 10 Build 19042
* Firefox version: Firefox 85.0.2 (64 bit)

tofuman0, 2021-03-09:
Try using the 64bit Webcamoid. Appears to work fine for me with 64bit apps. Equally I have to use the 32bit Webcamoid if I'm using skype for instance which is 32bit.

GaGag23, 2021-03-23:
Thanks, it worked. The 64-bit version crashes on start, but I edited share/config/Webcamoid.ini and added
`playOnStart=false` in [StreamConfig]
and changed the Video Capture Lib to mediafundation and it worked. Probably a dshow issue.

closed 2021-03-23 by GaGag23