webcamoid/webcamoid #408: Can't write header:  Error number 22 occurred

WilliamTambellini, 2021-02-20:
## Summary

Try to record an AVI using the laptop webcam, see the UI behaves just as everything alright, but the output file is totally wrong (attached)

## Current Behavior

The output file is 
`more ~/Videos/Video\ 2021-02-19\ 14-22-56.avi
 
RIFF����AVI LIST����hdrlavih8
`

## Expected Behavior

Webcamoid to at least warn on the UI that recording failed for these options.

## Steps to Reproduce

1.  Try to record an AVI with webcamoid 8.7.1, ubuntu 18
2.  Click 'Start recording', wait few secs
3.  Click Stop recording
4.  See the output file is only 160 bytes and in the console: 
`
Can't write header:  Error number 22 occurred
Error requesting latest version: No response 
`

## Suggestions and tips

1 adding more logs 2 Adding asserts 3 gdb

## Your Environment

* Webcamoid information: 8.7.1 64b
* Operating System information: 
Linux wtamb 4.15.0-1069-oem #79-Ubuntu SMP Wed Feb 5 13:48:18 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
* Any other useful information:  screenshots attached
Tks

[image]
[image]

sandsturm, 2021-03-22:
I have the same issue with mkv. But it works with h264_nvenc. Most of the other codecs result in the error number 22.

cyqsimon, 2021-09-17:
Same issue on my box (Manjaro x64). I tried `webm` and `mp4` output formats, using software and hardware (NV & Intel) codecs. All that I've tested resulted in error 22.

Depending on the output format, the output file was either empty (0B) or tiny (264B for webm). Not sure how this could help but just FYI I guess.

hipersayanX, 2022-02-15:
Problem solved in 9.0.0, closing.

chrneumann, 2023-03-23:
@hipersayanX Can't confirm that, I also get this Error 22 for version 9.0.0. Tried different formats. Pictures work.

chrneumann, 2023-03-23:
Works if I switch from ffmpeg to gstreamer for video recording. The next issue is that I can't change the resolution, it's always 640x480 (similar to #379). Using a Logitech Webcam (C920), almost as the guy in #379. Possible related to the hardware?

chrneumann, 2023-03-23:
I just checked, I recorded a Video with Webcamoid and this webcam about a year ago in 864x480 and there where no problems.

hipersayanX, 2023-03-25:
@chrneumann 

>Can't confirm that, I also get this Error 22 for version 9.0.0. Tried different formats. Pictures work.
>
>Also tried daily build, problem persists.

I was testing recording in Webm just a few days ago and it was working fine with FFmpeg.

> Works if I switch from ffmpeg to gstreamer for video recording. The next issue is that I can't change the resolution, it's always 640x480 (similar to [...]

Nope, Webcamoid takes the resolution of the source when you started recording and keep it until the end of the recording, if you started recording with a source at 640x480, and now you switch to a source with different resolution, it will keep recording at 640x480 until you stop the recording.

chrneumann, 2023-03-25:
> Nope, Webcamoid takes the resolution of the source when you started recording and keep it until the end of the recording, if you started recording with a source at 640x480, and now you switch to a source with different resolution, it will keep recording at 640x480 until you stop the recording.

I switch before recording (which changes the resolution of the preview stream, but not the recorded video).

hipersayanX, 2023-03-25:
>I switch before recording (which changes the resolution of the preview stream, but not the recorded video).

Ok, will check then.

chrneumann, 2023-03-25:
If you need any more help debugging, let me know.

hipersayanX, 2023-03-25:
>Also tried daily build, problem persists.

I was looking at it and it seems that the latest valid build was like 3 weeks ago, and I have fixed some bugs in the [video recording](https://github.com/webcamoid/webcamoid/commit/b5dfb2895b62ab0dc9e5807d6efa847868911df4) the past week, maybe the bug was fixed there, not sure. Right now I'm trying to add support for the virtual camera in Flatpak, and will check the builds later.

hipersayanX, 2023-06-10:
Bug fixed in Webcamoid 9.1.0.

closed 2023-06-10 by hipersayanX

GribanovIvan, 2024-08-31:
Still happens on 9.1.1.
default config, ArchLinux, tried to record video, Lenovo IdeaPad Gaming 3 15ARH7 on Ryzen 5 6600H.
Switching recorder to gstreamer did not help.
[Output file](https://github.com/user-attachments/assets/864ee970-f362-4a9d-84a1-007e7fbbf2b3)
Console:
```
[2024-08-31 13:18:31.813, Webcamoid, 0x70d47ed85880,  (0)] debug: Added component "Webcamoid"
[2024-08-31 13:18:31.813, Webcamoid, 0x70d47ed85880,  (0)] debug: Added component "VirtualCamera"
[matroska,webm @ 0x70d360c08ac0] Duplicate element
[matroska,webm @ 0x70d360c08ac0] 0x00 at pos 100 (0x64) invalid as first byte of an EBML number
[matroska,webm @ 0x70d360c08ac0] Duplicate element
[matroska,webm @ 0x70d360c08ac0] 0x00 at pos 169 (0xa9) invalid as first byte of an EBML number
[matroska,webm @ 0x70d360c08ac0] Element at 0x5d ending at 0x3fec0100000066 exceeds containing master element ending at 0x1409
[matroska,webm @ 0x70d360c08ac0] Duplicate element
[matroska,webm @ 0x70d360c08ac0] 0x00 at pos 100 (0x64) invalid as first byte of an EBML number
[matroska,webm @ 0x70d360c08ac0] Duplicate element
[matroska,webm @ 0x70d360c08ac0] 0x00 at pos 169 (0xa9) invalid as first byte of an EBML number
[matroska,webm @ 0x70d360c08ac0] Element at 0x5d ending at 0x3fec0100000066 exceeds containing master element ending at 0x1409
```