webcamoid/webcamoid #409: Webcam-video-recording with MP4 format is not working on Ubuntu.

YoungMahesh, 2021-02-24:
## Summary
When I record MP4 videos, the size of the file generated after video recording stopped, is of size - 0 bytes.

## Current Behavior
- When I record videos with 3GP2 (3GPP2 file format) or 3GP(3GPP file format) format, video file of expected size is generated (normal behavior)
- But when I record videos with MP4 (MPEG-4 Part 14) format, video file with size of **0 bytes** is generated (unexpected behavior)

## Expected Behavior
- When I record videos with MP4 format, the video file generated should be **> 0 bytes** of size.

## Steps to Reproduce
1.  Open WebCamoid .AppImage file
2.  Click on **Record Video** from dock at bottom-side of screen
3.  Select **MP4 (MPEG-4 Part 14)** from left-hand side column
4.  Click **Start recording video** on right-bottom corner, pop will came about asking file-name, click Enter
5. Wait for 10 seconds to record video
6. Click on **Stop recording video**
7. Go to Videos-section of file-manager `~/Videos`
8. The file generated as .mp4 format is of size 0 bytes.

## My Environment
* Webcamoid information: v8.8.0 (latest version), x64_86 architecure, .AppImage file
* Operating System information: Ubuntu 20.04 (Focal Fossa), AMD64 - desktop image
* Hardware: Dell vostro 15 3578

deegeebee, 2021-07-15:
Same here.

MP4 screencaptures with Kazaam work fine.

Environment:
Webcamoid: webcamoid-8.8.0-x86_64.AppImage
OS Ubuntu 20.04.2 with latest updates plus proprietary codec pack

Rajatisporwal, 2021-07-18:
Same here, recording with other file formats works smooth. The only exception is mp4

Environment:
Webcamoid: webcamoid-installer-8.8.0-x86_64.run
OS: Linux mint 20.04 xfce

fhavio, 2021-11-29:
Same bug with 8.6.1, appears that is recording and in the end file has 0 bytes,
Works with other formats

caltondoug, 2022-01-07:
I am having the same problem with Windows 10.  Recording works with 3GP, but nothing else I have tried.  All other formats result in 0k or 1k files.

hipersayanX, 2022-01-08:
@caltondoug check the daily build, it should be fixed there.

hipersayanX, 2022-02-15:
Problem solved in 9.0.0, closing.

Jeff-69, 2023-02-21:
same have problem
I just downloaded 9.0.0

bonzaiferroni, 2023-02-23:
I also have this problem with version 8.8.0 and 9.0.0

hipersayanX, 2023-06-10:
Bug fixed in Webcamoid 9.1.0.

closed 2023-06-10 by hipersayanX