webcamoid/webcamoid #410: Webcamoid fails to launch on Ubuntu 21.04 dev branch

mcl21014, 2021-02-28:
Choose one of the templates bellow that fit better your issue.

# Reporting a problem?

## Summary

Crash on startup.

## Current Behavior
Terminal output (After running installer and selecting Webcamoid to run):
mike@mike-CyberPowerPC:~/Downloads$ QApplication: invalid style override passed, ignoring it.
    Available styles: Windows, Fusion
QQmlApplicationEngine failed to load component
qrc:/Webcamoid/share/qml/main.qml:20 plugin cannot be loaded for module "QtQuick": The plugin '/usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick.2/libqtquick2plugin.so' uses incompatible Qt library. (5.15.0) [release]

The result is the application will not launch, as mentioned prior.
Describe the problem the best as you can, don't omit information.

## Expected Behavior
Webcamoid should start as usual.
## Steps to Reproduce
If Webcamoid crashed, try to reproduce the crash several times to be sure where is the problem. Write the steps to reproduce the issue bellow:

1.  Install Ubuntu 21.04 daily iso and update it fully.
2.  Download either appimage or installer version of Webcamoid stable or nightly. Grant permission to execute and run and install.
3.  Run Webcamoid from terminal post install, or run the binary from the installation folder.
4. 
## Suggestions and tips

I've deleted the old config files in ~./config/webcamoid and the whole .webcamoid folder as troubleshooting. This bug didn't happen on the 8.7.x version of Webcamoid, to my knowledge.
How would you solve the problem?

## Your Environment

* Webcamoid information: Webcamoid version 8.8.0 x86_64
* Operating System information: 
OS: Ubuntu 21.04 hirsute x86_64
* Any other useful information: (logs, gdb backtrace, valgrind logs, screenshots, hardware, etc.)
Nothing interesting in the install log from what I've seen. Please give feedback if more logs are needed.

# Questinons and other matters?
None.

hipersayanX, 2021-03-01:
It's trying to load the Qt plugins from your system instead of the bundle,, its most probably a bug in the deploy, I'm working on it right now.

mcl21014, 2021-03-01:
Good job getting on it so early, and your work is appreciated. I hear feedback not even a day after posting this. What you say must be true as the other complaint was from Ubuntu 20.04 LTS. No way a dev release and LTS share the same exact versions of those QT plugins.

hipersayanX, 2021-03-02:
I've started working on it a few days ago, I was planning to port all the build system to cmake instead of qmake and the deploy scripts did not handled the case very well.

mrtwiste, 2021-03-26:
Same issue:
NAME="Ubuntu" VERSION="20.10 (Groovy Gorilla)"
QQmlApplicationEngine failed to load component
qrc:/Webcamoid/share/qml/main.qml:20 plugin cannot be loaded for module "QtQuick": The plugin '/usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick.2/libqtquick2plugin.so' uses incompatible Qt library. (5.14.0) [release]

denialmedia, 2021-03-30:
same issue. Pop
Distributor ID:	Pop
Description:	Pop!_OS 20.10
Release:	20.10
Codename:	groovy

QQmlApplicationEngine failed to load component
qrc:/Webcamoid/share/qml/main.qml:20 plugin cannot be loaded for module "QtQuick": The plugin '/usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick.2/libqtquick2plugin.so' uses incompatible Qt library. (5.14.0) [release]

hipersayanX, 2022-02-15:
Problem solved in 9.0.0, closing.

closed 2022-02-15 by hipersayanX