webcamoid/webcamoid #411: Regression: appimage of webcamoid 8.8.0 broken (hangs on start) on ubuntu 20.04

callegar, 2021-03-01:
## Summary

Appimage of webcamoid 8.8.0 does not work on ubuntu 20.04 due to library conflicts. It hangs on start.
Appimage of webcamoid 8.7.1 works fine.

## Current Behavior

Trying to launch the appimage of webcamoid 8.8.0 reports
```
QQmlApplicationEngine failed to load component
qrc:/Webcamoid/share/qml/main.qml:24 plugin cannot be loaded for module "AkQml": Cannot load library /usr/lib/x86_64-linux-gnu/qt5/qml/AkQml/libAkQml.so: (/usr/lib/x86_64-linux-gnu/qt5/qml/AkQml/libAkQml.so: undefined symbol: _ZN9AkElementclERK13AkAudioPacket)
```
and the application hangs with no GUI.

## Expected Behavior

Webcamoid 8.8.0 shoud start regularly, just like previous 8.7.1

## Steps to Reproduce

1. Try to start the application on ubuntu 20.04

## Suggestions and tips

None

## Your Environment

* Webcamoid information: 8.8.0 x86_64 applimage
* Operating System information: ubuntu 20.04 (in fact kubuntu) x86_64

leogama, 2021-03-19:
Hi! The 8.8.0 appimage worked flawlessly for me on Ubuntu 20.04. However, today I installed Webcamoid from Ubuntu's repositories using APT (version 8.6.1) and, *after this*, I started getting this same error with the appimage. Here is the complete list of packages installed alongside `webcamoid`:

`akqml libavkys8 libqt5multimedia5 libqt5quickcontrols2-5 libqt5quicktemplates2-5 qml-module-qt-labs-folderlistmodel qml-module-qt-labs-settings qml-module-qtqml-models2 qml-module-qtquick-controls2 qml-module-qtquick-dialogs qml-module-qtquick-extras qml-module-qtquick-privatewidgets qml-module-qtquick-templates2 webcamoid webcamoid-data webcamoid-plugins`

I didn't try to uninstall these and rerun the appimage yet to see if it works back, but I think that, for some reason, the appimage uses the system's Qt QML library if it is available instead of its own.

hipersayanX, 2021-03-19:
@leogama try with [this](https://github.com/webcamoid/webcamoid/wiki/Reset-configs-and-starting-fresh#gnulinux).

FM33, 2021-03-19:
Same problem here. Even after config reset, it doesn't work. Here is the output:
```
fabien@fabien-desktop:~/Téléchargements$ ./webcamoid-8.8.0-x86_64.AppImage 
QQmlApplicationEngine failed to load component
qrc:/Webcamoid/share/qml/main.qml:24 plugin cannot be loaded for module "AkQml": Cannot load library /usr/lib/x86_64-linux-gnu/qt5/qml/AkQml/libAkQml.so: (/usr/lib/x86_64-linux-gnu/qt5/qml/AkQml/libAkQml.so: undefined symbol: _ZN9AkElementclERK13AkAudioPacket)
```

I have akqml 8.6.1+dfsg-2build1 installed

callegar, 2021-04-07:
Triaged. Issue happens when you also have webcamoid installed from the ubuntu 20.04 deb package. For some reason the webcamoid 8.8.0 appimage seems to prioritize the AkQml lib that is pulled in by the webcamoid deb, and that does not work with it.

ghousekhas, 2021-05-31:
Any temporary fixes or workarounds? Have the same problem.

hipersayanX, 2021-06-02:
@ghousekhas check if you have the following lines in the _bin/qt.conf_ file:

```
[Paths]
Plugins = ../lib/qt/plugins
Imports = ../lib/qt/qml
Qml2Imports = ../lib/qt/qml
```

or try giving the absolute paths, there are not other solutions for now.

hipersayanX, 2022-02-15:
Problem solved in 9.0.0, closing.

closed 2022-02-15 by hipersayanX