webcamoid/webcamoid #419: Webcamoid doesn't start on Win 10 but just appears in background processes.

LumenLuminis, 2021-03-20:
Current Behavior:

Webcamoid doesnt open (in any download) but just sits around in Background Processes, Loading about 20 MBs to ram and consuming small amounts of processor power. Possibly connected to broken build.
Version:
Latest Github Release (8.8.0) and daily builds.

knw257, 2021-07-10:
I'm also experiencing this issue, version 8.8.0. Webcamoid.exe appears in task manager, but produces no window. Can end process and re-run multiple times, no change. Have deleted HKCU\SOFTWARE\Webcamoid with no change. Have unplugged all webcams from system, no change. I also have OBS and Streamlabs OBS installed, both of which have virtual webcams, but I'm not sure how to remove those without completely uninstalling the software, which I use for streaming.

brossow, 2021-08-11:
I get something similar. The app window opens half on and half off the screen for a moment, then disappears from the screen and Task Manager. Windows 10 with all updates, latest version of Webcamoid, some generic USB webcam that works fine with everything else.

hipersayanX, 2022-02-15:
Problem solved in 9.0.0, closing.

closed 2022-02-15 by hipersayanX