webcamoid/webcamoid #422: Immediate crash and can't uninstall

rycky000, 2021-03-24:
Windows 10 64 bit, latest release
It crashes immediately after the start, No error notification, no "webcamoid.exe has stopped working" or something else... The webcamoid window appear for 1 second and then it just disappear. So I tried to uninstall but i just find out that i can't. Double clicking WebcamoidUninstall.exe or try to uninstall using Control Panel doesn't have any sort of effect.
Tried to reboot, to uninstall in safeboot, to use Geek uninstaller... Nothing

Edit: After 1 hour it started working but is super slow and eventually crash, I cant get my webcam working with the application even if it works with all the other programs on my pc. And i still can't uninstall.

hipersayanX, 2021-03-25:
>And i still can't uninstall.

Try with [this](https://support.microsoft.com/en-us/topic/how-to-manually-remove-programs-from-the-add-remove-programs-list-49494159-d215-07b2-e4c7-050457b38352).
Its apparently a bug introduced in the latest versions of Qt IFW, I'll have to switch to another install method.

rycky000, 2021-03-25:
> Try with [this](https://support.microsoft.com/en-us/topic/how-to-manually-remove-programs-from-the-add-remove-programs-list-49494159-d215-07b2-e4c7-050457b38352).
> Its apparently a bug introduced in [...]
I can't find a key with a "webcamoid"-like display name in the regedit. But the program is still in the Control Panel and i can still start it (even if most of the times it crashes).

rycky000, 2021-03-25:
Ok i uninstalled webcamoid using Revo uninstaller. The program was still broken. But i realized that with webcam off the program starts without crashing immediately and it work (even if super slow and it sometime freezes). But if i turn on the webcam (before or while webcamoid is running) and i select it as a source, webcamoid immediately crashes.
My pc:
Windows 10 Pro 64 bit
Ryzen 3600 and radeon rx5700xt (amd graphics drivers up to date)
Webcam Aukey "2K" 30fps

DoubleBullet, 2021-03-26:
I uninstalled it easily by going to Control Panel\Programs\Programs and Features then clicking uninstall on webcamoid.

rycky000, 2021-03-26:
> 
> 
> I uninstalled it easily by going to Control Panel\Programs\Programs and Features then clicking uninstall on webcamoid.

Yeah I know how to uninstall a program but as I've already said that wasn't working with webcamoid.

hipersayanX, 2021-03-26:
@DoubleBullet isn't a problem of no knowing how to uninstall the program, this is an actual bug of the installer itself, QtIFW introduced this bug in the latest releases, and sadly this bug affected the latest release and the daily build.

YorVeX, 2021-05-19:
Same problem here, it doesn't start, not even after a reboot and there's not error message, just the window being visible for a moment and then gone again (same for the webcamoid process on the task manager). Windows 10 64 bit 20H2 here. There doesn't seem to be any log either, the only thing I could find was an entry in Windows event log mentioning swscale-5.dll to be the faulty module that caused webcamoid.exe to fail. Exception code 0xc0000005 and error offset 0x0000000000066e81 if that helps.

At least uninstalling worked fine afterwards.

hipersayanX, 2022-02-15:
Problem solved in 9.0.0, closing.

closed 2022-02-15 by hipersayanX