webcamoid/webcamoid #428: (win) After uninstall I still get: "The Webcamoid virtual camera service service terminated unexpectedly..."

samozrejme, 2021-04-10:
After I've uninstall Webcamoid I still get Errors in the Even Viewer on Windows.

Event Viewer - Windows Logs - System:
_The Webcamoid virtual camera service service terminated unexpectedly.  It has done this 13 time(s).  The following corrective action will be taken in 5000 milliseconds: Restart the service._

This error appears exactly every 30 minutes. The number in "It has done this _n_ time(s)." changes a bit. My system freezes for a second during this event.

I have also tried reinstalling the current version of Webcamoid and unistalling it with the uninstall.exe, but the error is still happening.

Please help me remove all of the Webcamoid files of my system.
And maybe please make sure the uninstall tool does its job as it is probably supposed to.
Thx

hipersayanX, 2021-04-11:
>Please help me remove all of the Webcamoid files of my system.

Just click **Remove All** to remove the virtual camera, that should fix the problem.

[image]

>And maybe please make sure the uninstall tool does its job as it is probably supposed to.

This bug will be fixed in Webcamoid 9.

samozrejme, 2021-04-12:
> > Please help me remove all of the Webcamoid files of my system.
> 
> Just click **Remove All** to remove the virtual camera, that should fix the problem.

Ok, had to reinstall it again to do so, but it seems to have fixed the issue. Thanks for your help.

samozrejme, 2021-04-12:
Not sure if I should keep this open or not.
Since I got my answer but the issue can still occur for other users.

(github noob)

hipersayanX, 2021-04-16:
>Not sure if I should keep this open or not.
>Since I got my answer but the issue can still occur for other users.

Yes, that's the right way since the bug is still ongoing.

kianboon80, 2021-06-10:
i am also facing the same issue with portable version. How can i uninstall? I delete the folder away but problem still exist?

hipersayanX, 2021-06-10:
@kianboon80 just do as said [here](https://github.com/webcamoid/webcamoid/issues/428#issuecomment-817238532).

kianboon80, 2021-06-10:
@hipersayanX thank for your advise but I am unable to follow the instruction. When I open the app, it seems to initaite the laptop built-in cam and will shut down shortly after that.whole process in less than 2s. I am unable to access the app.

hipersayanX, 2021-06-10:
@kianboon80 try [resetting Webcamoid](https://github.com/webcamoid/webcamoid/wiki/Reset-configs-and-starting-fresh#windows) and try again, if that does not work you will need to uninstall the virtual camera service manually.
Open an administrator cmd and run the following commands:

```
cd "C:\Program Files\AkVirtualCamera.plugin"
sc start AkVCamAssistant
regsvr32 /s /u .\x86\AkVirtualCamera.dll
regsvr32 /s /u .\x64\AkVirtualCamera.dll
sc stop AkVCamAssistant
.\x86\AkVCamAssistant  --uninstall
.\x64\AkVCamAssistant  --uninstall
```

hipersayanX, 2022-02-15:
Problem solved in 9.0.0, closing.

closed 2022-02-15 by hipersayanX