webcamoid/webcamoid #431: [question] Can webcamoid + akvirtualcamera create virtual backgrounds for zoom?

kirk86, 2021-04-20:
Sorry if this sounds like a stupid question but I was wondering if one could use webcamoid + akvirtualcamera to alter or create virtual backgrounds to hide your background, similar to what zoom offers and then feed that into zoom or other conferencing app?

You see, zoom has this feature but unfortunately doesn't work with my old macbook 2015 without a green screen.

hipersayanX, 2021-04-21:
Virtual backgrousds has not been implemented yet, there is an issue for [that feature](https://github.com/webcamoid/webcamoid/issues/250) already.
Also, I've not tested the virtual camera with Zoom yet.

kirk86, 2021-04-21:
Thanks for letting me know!

cont1nuity, 2021-05-03:
> 
> 
> Virtual backgrousds has not been implemented yet, there is an issue for [that feature](https://github.com/webcamoid/webcamoid/issues/250) already.
> Also, I've not tested the virtual camera with Zoom yet.

What about https://github.com/webcamoid/webcamoid/pull/425? That is not exactly like in Zoom, but works the same. Not yet released, though.

hipersayanX, 2021-05-03:
@cont1nuity isn't exactly the same, what we want is something like [this](https://github.com/webcamoid/webcamoid/issues/250#issuecomment-660695722), the point is replacing the static background while living the moving objects in the front.
There is a lot of algorithms and codes for doing so, but I had not time to dig into it, I'm more focused on releasing webcamoid 9 once for all.

cont1nuity, 2021-05-03:
@hipersayanX True. #425 and #430 are just the "cheap" solution for this :)

hipersayanX, 2021-05-03:
@cont1nuity but still a good job, don't demerit your work :wink:

kirk86, 2021-05-03:
@hipersayanX @hipersayanX So if I understand correctly #425 and #430 are not doing this?

> the point is replacing the static background while living the moving objects in the front.

My other question is even if their are not perfect will we be able to feed to virtual camera in order to show in zoom?

cont1nuity, 2021-05-03:
> 
> 
> @hipersayanX @hipersayanX So if I understand correctly #425 and #430 are not doing this?

They are, but you need to "cut" your face a bit. The detection will work and the area will move with your face.
> 
> > the point is replacing the static background while living the moving objects in the front.
> 
> My other question is even if their are not perfect will we be able to feed to virtual camera in order to show in zoom?

Sorry, no clue, but I assume it should work as virtual camera source.

closed 2022-02-15 by hipersayanX