webcamoid/webcamoid #433: webcamoid hangs with Jabra PanaCast

joakim-tjernlund, 2021-05-12:
# README

webcamoid 8.7.1 just hangs whenever above camera is connected. 

```
 gst-device-monitor-1.0 Video/Source
Probing devices...

Device found:

	name  : Jabra PanaCast
	class : Video/Source
	caps  : image/jpeg, width=(int)3840, height=(int)1080, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 22/1, 15/1 };
	        image/jpeg, width=(int)1920, height=(int)1080, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 15/1 };
	        image/jpeg, width=(int)1280, height=(int)720, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 27/1, 24/1, 15/1, 30/1, 27/1, 24/1, 15/1, 30/1, 27/1, 24/1, 15/1 };
	        image/jpeg, width=(int)1280, height=(int)720, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 27/1, 24/1, 15/1, 30/1, 27/1, 24/1, 15/1, 30/1, 27/1, 24/1, 15/1 };
	        image/jpeg, width=(int)960, height=(int)544, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 27/1, 24/1, 15/1 };
	        image/jpeg, width=(int)640, height=(int)360, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 27/1, 24/1, 15/1 };
	properties:
		udev-probed = true
		device.bus_path = pci-0000:00:14.0-usb-0:2:1.0
		sysfs.path = /sys/devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.0/video4linux/video0
		device.bus = usb
		device.subsystem = video4linux
		device.vendor.id = 2b93
		device.vendor.name = "GN\\x20Jabra\\x20A\\x2fS"
		device.product.id = 0003
		device.product.name = "Jabra\ PanaCast"
		device.serial = GN_Jabra_A_S_Jabra_PanaCast_00212442774
		device.capabilities = :capture:
		device.api = v4l2
		device.path = /dev/video0
		v4l2.device.driver = uvcvideo
		v4l2.device.card = "Jabra\ PanaCast"
		v4l2.device.bus_info = usb-0000:00:14.0-2
		v4l2.device.version = 330275 (0x00050a23)
		v4l2.device.capabilities = 2225078273 (0x84a00001)
		v4l2.device.device_caps = 69206017 (0x04200001)
	gst-launch-1.0 v4l2src ! ...
```

Might it be because  the camera just have caps image/jpeg ?

Camera works fine in MS Teams and guvcview but not cheese

gstreamer 1.16.3 also has a problem:
```
gst-launch-1.0 -e --gst-debug-level=10 v4l2src device=/dev/video0 ! videoconvert
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
ERROR: from element /GstPipeline:pipeline0/GstV4l2Src:v4l2src0: Internal data stream error.
Additional debug info:
/var/tmp/portage/media-libs/gstreamer-1.16.3/work/gstreamer-1.16.3/libs/gst/base/gstbasesrc.c(3072): gst_base_src_loop (): /GstPipeline:pipeline0/GstV4l2Src:v4l2src0:
streaming stopped, reason not-negotiated (-4)
EOS on shutdown enabled -- waiting for EOS after Error
Waiting for EOS...
```

hipersayanX, 2021-05-13:
What is the output of this?

```
v4l2-ctl -d /dev/video0 --all -L
v4l2-compliance -d /dev/video0 -s -f 
```

joakim-tjernlund, 2021-05-15:
been traveling but here goes:
```
v4l2-ctl -d /dev/video0 --all -L
Driver Info:
	Driver name      : uvcvideo
	Card type        : Jabra PanaCast
	Bus info         : usb-0000:00:14.0-2
	Driver version   : 5.10.35
	Capabilities     : 0x84a00001
		Video Capture
		Metadata Capture
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps      : 0x04200001
		Video Capture
		Streaming
		Extended Pix Format
Media Driver Info:
	Driver name      : uvcvideo
	Model            : Jabra PanaCast
	Serial           : 00212442774
	Bus info         : usb-0000:00:14.0-2
	Media version    : 5.10.35
	Hardware revision: 0x00000002 (2)
	Driver version   : 5.10.35
Interface Info:
	ID               : 0x03000002
	Type             : V4L Video
Entity Info:
	ID               : 0x00000001 (1)
	Name             : Jabra PanaCast
	Function         : V4L2 I/O
	Flags         : default
	Pad 0x01000007   : 0: Sink
	  Link 0x02000010: from remote pad 0x100000a of entity 'Extension 3': Data, Enabled, Immutable
Priority: 2
Video input : 0 (Camera 1: ok)
Format Video Capture:
	Width/Height      : 1920/1080
	Pixel Format      : 'MJPG' (Motion-JPEG)
	Field             : None
	Bytes per Line    : 0
	Size Image        : 4147200
	Colorspace        : sRGB
	Transfer Function : Rec. 709
	YCbCr/HSV Encoding: ITU-R 601
	Quantization      : Default (maps to Full Range)
	Flags             : 
Crop Capability Video Capture:
	Bounds      : Left 0, Top 0, Width 1920, Height 1080
	Default     : Left 0, Top 0, Width 1920, Height 1080
	Pixel Aspect: 1/1
Selection Video Capture: crop_default, Left 0, Top 0, Width 1920, Height 1080, Flags: 
Selection Video Capture: crop_bounds, Left 0, Top 0, Width 1920, Height 1080, Flags: 
Streaming Parameters Video Capture:
	Capabilities     : timeperframe
	Frames per second: 15.000 (15/1)
	Read buffers     : 0
                     brightness 0x00980900 (int)    : min=0 max=255 step=1 default=145 value=145
                       contrast 0x00980901 (int)    : min=0 max=191 step=1 default=110 value=110
                     saturation 0x00980902 (int)    : min=128 max=176 step=1 default=143 value=143
 white_balance_temperature_auto 0x0098090c (bool)   : default=1 value=1
                           gain 0x00980913 (int)    : min=0 max=255 step=1 default=229 value=229
      white_balance_temperature 0x0098091a (int)    : min=0 max=6500 step=1 default=3700 value=3700 flags=inactive
                      sharpness 0x0098091b (int)    : min=0 max=255 step=1 default=144 value=144
         backlight_compensation 0x0098091c (int)    : min=0 max=255 step=1 default=229 value=229
                   pan_absolute 0x009a0908 (int)    : min=-302400 max=302400 step=1 default=0 value=0
                  tilt_absolute 0x009a0909 (int)    : min=-72000 max=72000 step=1 default=0 value=0
                  zoom_absolute 0x009a090d (int)    : min=1 max=75 step=1 default=1 value=1
[... 193 lines cut ...]
	test MMAP for Format MJPG, Frame Size 1280x720@15.00 Hz:
		Stride 0, Field None: OK                                    
	test MMAP for Format MJPG, Frame Size 1280x720@30.00 Hz:
		Stride 0, Field None: OK                                    
	test MMAP for Format MJPG, Frame Size 1280x720@27.00 Hz:
		Stride 0, Field None: OK                                    
	test MMAP for Format MJPG, Frame Size 1280x720@24.00 Hz:
		Stride 0, Field None: OK                                    
	test MMAP for Format MJPG, Frame Size 1280x720@15.00 Hz:
		Stride 0, Field None: OK                                    
	test MMAP for Format MJPG, Frame Size 1280x720@30.00 Hz:
		Stride 0, Field None: OK                                    
	test MMAP for Format MJPG, Frame Size 1280x720@27.00 Hz:
		Stride 0, Field None: OK                                    
	test MMAP for Format MJPG, Frame Size 1280x720@24.00 Hz:
		Stride 0, Field None: OK                                    
	test MMAP for Format MJPG, Frame Size 1280x720@15.00 Hz:
		Stride 0, Field None: OK                                    
	test MMAP for Format MJPG, Frame Size 3840x1080@22.00 Hz:
		Stride 0, Field None: OK                                    
	test MMAP for Format MJPG, Frame Size 3840x1080@15.00 Hz:
		Stride 0, Field None: OK                                    
Total for uvcvideo device /dev/video0: 90, Succeeded: 82, Failed: 8, Warnings: 5
```
Seems to be a few fails there.

hipersayanX, 2021-05-16:
I think these are the two conflicting lines:

```
test VIDIOC_ENUM_FMT/FRAMESIZES/FRAMEINTERVALS: OK
    warn: v4l2-test-formats.cpp(1339): S_PARM is supported but doesn't report V4L2_CAP_TIMEPERFRAME
    fail: v4l2-test-formats.cpp(1241): node->has_frmintervals && !cap->capability
test VIDIOC_TRY_FMT: OK
    warn: v4l2-test-formats.cpp(988): Could not set fmt2
```

It seems that isn't possible to get or set the frame rate and the format of that camera. The frame rate problem was already solved long time ago in the master branch, the second will need some tweak.

joakim-tjernlund, 2021-05-18:
FYI, I am on Gentoo and can build from master so shout when you think you have something I can test.

joakim-tjernlund, 2021-05-18:
BTW, I have a builtin webcam, ID 04f2:b6d0 Chicony Electronics Co., Ltd Integrated Camera, too.
This Cam works and has the 1:st fail but not the 2:nd so your idea seem spot on.

However, this Cam spews a lot of( via ffmpeg ):
    av_log(c, AV_LOG_WARNING, "deprecated pixel format used, make sure you did set range correctly\n");

This comes from libswscale/utils.c in ffmpeg:
```
     if(srcFormat!=c->srcFormat || dstFormat!=c->dstFormat)
         av_log(c, AV_LOG_WARNING, "deprecated pixel format used, make sure you did set range correctly\n");
```
Any chance webcamoid can do anything about this?

hipersayanX, 2021-05-19:
The "deprecated pixel format" is a very common message in ffmpeg, nothing to worry about.

>Any chance webcamoid can do anything about this?

Sure, but I'll check it much later, I'm busy with other things right now.

joakim-tjernlund, 2021-05-19:
> The "deprecated pixel format" is a very common message in ffmpeg, nothing to worry about.

I know, it is just that the msg floods the terminal.
> 
> > Any chance webcamoid can do anything about this?
> 
> Sure, but I'll check it much later, I'm busy with other things right now.

Thanks, there is no hurry with this one.

joakim-tjernlund, 2021-07-11:
> I think these are the two conflicting lines:
>
> ```
> test VIDIOC_ENUM_FMT/FRAMESIZES/FRAMEINTERVALS: OK
>     warn: v4l2-test-formats.cpp(1339): S_PARM is supported but doesn't report [...]

Any progress on this issue?

hipersayanX, 2021-07-11:
@joakim-tjernlund nope, I'm busy with other things right now, I won't come back to this issue until at least some months or years maybe, but there is the [source code](https://github.com/webcamoid/webcamoid) for whoever want to collaborate fixing this.

hipersayanX, 2022-02-15:
Problem solved in 9.0.0, closing.

closed 2022-02-15 by hipersayanX