webcamoid/webcamoid #434: Webcamoid doesn't find virtual camera driver

ahmedkhalf, 2021-05-25:
I downloaded both v4l2loopback-dkms and akvcam, but there is no item in the dropdown list for virtual cam drivers.

* Webcamoid information: Webcamoid 8.8.0
* Operating System information: 5.10.36-2-MANJARO

hipersayanX, 2021-05-26:
What is the result of this command?

```
cat /lib/modules/$(uname -r)/modules.dep | grep akvcam
```

ahmedkhalf, 2021-05-26:
Nothing is returned except for a non zero return code, meaning it wasn't able to find it.

hipersayanX, 2021-05-26:
The driver was not installed properly, reinstall akvcam-dkms from AUR.

ahmedkhalf, 2021-05-26:
I reinstalled, still same problem, I even tried both the git and the normal one...

hipersayanX, 2021-05-26:
Does these lines (or similar ones) appears at the end when installing from AUR?

```
:: Running pre-transaction hooks...
(1/1) Remove DKMS modules
==> dkms remove --no-depmod -m akvcam -v 1.2.0.r1.g3d7690e -k 5.12.5-arch1-1
==> depmod 5.12.5-arch1-1
:: Processing package changes...
(1/1) upgrading akvcam-dkms-git                                                         [###################################################] 100%
:: Running post-transaction hooks...
(1/2) Arming ConditionNeedsUpdate...
(2/2) Install DKMS modules
==> dkms install --no-depmod -m akvcam -v 1.2.0.r14.gf6d9dd3 -k 5.12.5-arch1-1
==> depmod 5.12.5-arch1-1
No database errors have been found!
```

ahmedkhalf, 2021-05-26:
Also getting a similar error for v4l2
```
Preparing...
Synchronizing package databases...
Resolving dependencies...
Checking inter-conflicts...
Checking keyring...
Checking integrity...
Loading packages files...
Checking file conflicts...
Checking available disk space...
Installing v4l2loopback-dkms (0.12.5-1)...
Running post-transaction hooks...
Arming ConditionNeedsUpdate...
Install DKMS modules
==> Unable to install module v4l2loopback/0.12.5 for kernel 5.10.36-2-MANJARO: Missing kernel headers.
Transaction successfully finished.
```

ahmedkhalf, 2021-05-26:
Installed the linux header package, everything works now.

closed 2021-05-26 by ahmedkhalf

hipersayanX, 2021-05-26:
>Installed the linux header package, everything works now.

Oh, I was about to point your were missing linux-headers, glad you have fixed it :smile: