webcamoid/webcamoid #435: Error creating camera (Mac OS)

TheSeamau5, 2021-05-29:
Hi, 

I'm on Mac OS 10.15.7 and I'm unable to create a Virtual camera. I opened the webcamoid app, clicked on preferences, and then in the Output tab, I checked Virtual camera and tried to "Add" a virtual camera. 

I get a prompt to enter my credentials
[image]

After successfully entering them, I simply see "Error creating camera" written. The camera was not created and there are no further error messages. 
[image]

Would love some more guidance on what is happening and how to resolve and/or assist.

Thanks

hipersayanX, 2021-05-30:
This is the first time for your using the virtual camera? maybe you have used an older version of Webcamoid? or it's a fresh install?

Leemusa, 2021-12-24:
I am also experiencing this error on Mac OS 10.14.6. I start the program but there is no way to create a virtual camera.

TheSeamau5, 2021-12-25:
@hipersayanX I had just installed it and then tried to create the virtual camera. It was my first time using it/fresh install. I haven't tried it since.

hipersayanX, 2021-12-27:
Try installing the daily build, I've improved the error messages and the virtual camera installation.

hipersayanX, 2022-02-15:
Problem solved in 9.0.0, closing.

closed 2022-02-15 by hipersayanX