webcamoid/webcamoid #436: since upgrade Fedora 34 not possible recording in no format

Nasdrovia69, 2021-06-06:
## Your Environment

Webcamoid 8.8.0 Qt 5.12.8
Fedora 34 Kde
# Questinons and other matters?

when i recording in avi format vlc report VLC ne peut pas décoder le format « AMVF » (No description for this codec) the file is well created and the preview is active
when i recording in webm format  the file stagnant 332 o regardless of the duration of the recording
when i recording in mp4 format the file stagnant 0 o
when i recording in Ogg the file is well created i can read with vlc but not sound     
Thanks if you can resolve problem i like webcamoid

hipersayanX, 2022-02-15:
Problem solved in 9.0.0, closing.

closed 2022-02-15 by hipersayanX

Nasdrovia69, 2022-02-15:
Thanks, it's working nice,    great work !!!