webcamoid/webcamoid #437: Are the build documentation out of date?

fangfufu, 2021-06-09:
This [webpage](https://github.com/webcamoid/webcamoid/wiki/Build-and-install-in-GNU-Linux-and-FreeBSD) suggests that I "open ``Webcamoid.pro`` file with Qt creator". However the file _Webcamoid.pro_ does not exist in the repository, but ``webcamoid.kdev4`` does, which can be opened using ``kdevelop``. Is this piece of information out of date?

hipersayanX, 2021-06-10:
Yes, I've switched to CMake a few months ago, so the Qmake build steps doesn't apply anymore. You can open the _CMakeLists.txt_ with Qt Creator, or use KDevelop, or VS Code, or XCode, or any other IDEs supporting CMake, now there are more IDE options for developers :smile:

fangfufu, 2021-06-10:
Thanks for the clarification. Feel free to close the issue. (Or leave it open as a reminder for yourself. :stuck_out_tongue:

hipersayanX, 2021-06-10:
Will leave open for now, I'm pretty sure more people will ask for the same.

bkw777, 2021-07-22:
"I've switched to CMake a few months ago, so the Qmake build steps doesn't apply anymore. You can open the CMakeLists.txt with Qt Creator, or use KDevelop, or VS Code, or XCode, or any other IDEs supporting CMake, now there are more IDE options for developers"

Ok, but, kdevelop 5.6.2 when I try to open project, it doesn't show any files at all. CmakeLists.txt is really there but I can't find any way to open or import it in kdevelop other than simply opening the file as a text file in an editor tab.

I tried QtCreator and that seemed to import the CmakeLists.txt and create a project. I tried to build the project for realease and of course got nothing but errors:
```/home/bkw/src/webcamoid/libAvKys/Plugins/ACapsConvert/src/ffmpegav/src/convertaudioffmpegav.cpp:127: warning: ‘AVAudioResampleContext* avresample_alloc_context()’ is deprecated [-Wdeprecated-declarations]
/home/bkw/src/webcamoid/libAvKys/Plugins/ACapsConvert/src/ffmpegav/src/convertaudioffmpegav.cpp: In member function ‘virtual bool ConvertAudioFFmpegAV::init(const AkAudioCaps&)’:
/home/bkw/src/webcamoid/libAvKys/Plugins/ACapsConvert/src/ffmpegav/src/convertaudioffmpegav.cpp:127:59: warning: ‘AVAudioResampleContext* avresample_alloc_context()’ is deprecated [-Wdeprecated-declarations]
  127 |     this->d->m_resampleContext = avresample_alloc_context();
      |                                                           ^
```
I did install the build deps from the ci/ubuntu-linux/install-deps.sh (though I did not execute that script, it doesn't look good to run outside of a throw-away build container)

Isn't there a plain make or cmake command that can be used without relying on any fancy but always-broken IDE?

hipersayanX, 2021-07-22:
@bkw777 

>I tried to build the project for realease and of course got nothing but errors:

Those are warnings, not errors, and can be ignored. If the build  reached to the end then you can try launching.

>Isn't there a plain make or cmake command that can be used without relying on any fancy but always-broken IDE?

Yes

```
cd webcamoid
cmake -S .  -B ../webcamoid-build
make -C ../webcamoid-build
```

bkw777, 2021-07-22:
> @bkw777
>
>
> Those are warnings, not errors, and can be ignored. If the build reached to the end then you can try launching.
>
>
> Yes
>
> ```
> cd webcamoid
> cmake -S .  -B ../webcamoid-build
> make -C [...]

Thank you!

hipersayanX, 2021-07-22:
@bkw777 Also, you can use cmake-gui for configuring the build

https://cmake.org/cmake/help/latest/guide/user-interaction/index.html
https://cmake.org/runningcmake/

closed 2022-02-15 by hipersayanX