webcamoid/webcamoid #439: How do I set custom Qt path in Kdevelop?

fangfufu, 2021-06-10:
I am on Debian, and my Qt is outdated. I had to download a copy from Qt website. I have no idea how to set custom Qt path in Kdevelop. I am actively looking this up on Google, but I think it might be quicker to ask you. Either way, I think this question might help others who want to contribute to this project, and I reckon the answer to this question should go into the Wiki. 

May I also ask, which distribution / operating system are you developing Webcamoid on?

fangfufu, 2021-06-10:
Basically I had to set the Path in Kdevelop's preference, as well as the location of cmake binary.

fangfufu, 2021-06-10:
I get a bunch of compilation errors like these:
```
/home/fangfufu/projects/webcamoid/StandAlone/share/qml/VideoCodecOptions.qml:37:14: error: Expected token `;'
        for (let i = mainLayout.children.length - 1; i >= startChildren; i--)
```

Is there any ways to contact you to get help? I don't know if commenting on this issue is the most appropriate method.

hipersayanX, 2021-06-11:
>I am on Debian Buster (Debian Stable), and my Qt is outdated. I had to download a copy from Qt website. I have no idea how to >set custom Qt path in Kdevelop. I am actively looking this up on Google, but I think it might be quicker to ask you.

Try setting the environment variables before opening Kdevelop, like this:

```
export QTDIR=/opt/Qt
export PATH=$QTDIR/bin:$PATH
export LD_LIBRARY_PATH=$QTDIR/lib:$LD_LIBRARY_PATH
export PKG_CONFIG_PATH=$QTDIR/lib/pkgconfig:$PKG_CONFIG_PATH
```

**QTDIR** is the path where Qt was installed.
Also remember that you can still use Qt Creator for developing Webcamoid, you add the Qt version you have compiled to the kits, and Qt Creator will set all those variables for you.

>May I also ask, which distribution / operating system are you developing Webcamoid on?

- Arch (my main working setup)
- Ubuntu (the latest stable version)
- Manjaro (when I need to test drivers)
- Windows 10
- Mac Catalina
- And I have a VM with Trident (ex TrueOS, ex FreeBSD) but it seems to be dead

>I get a bunch of compilation errors like these:

It's probably detecting the system Qt version instead of the newer one.

>Is there any ways to contact you to get help? I don't know if commenting on this issue is the most appropriate method.

I usually prefer this kind of issues to be public as it serve as reference for other people. So resolving issues here is perfectly fine.

bkw777, 2021-07-22:
So, it's Kdevelop? The building directions wiki says to load Webcamoid.pro file with Qt creator , but there is no such file.

hipersayanX, 2021-07-22:
@bkw777

>So, it's Kdevelop?

Qt Creator,  Kdevelop, CodeBlocks, CodeLite, Eclipse, Visual Studio, Xcode, and basically any IDE that supports Cmake.

https://cmake.org/cmake/help/git-master/manual/cmake-generators.7.html
https://en.wikipedia.org/wiki/Comparison_of_integrated_development_environments#C/C++

You are free to use any IDE you want. In my case I use Qt Creator because is the easiest to use for me.

>The building directions wiki says to load Webcamoid.pro file with Qt creator , but there is no such file.

The wiki was documented for Webcamoid 8.x which was build using Qmake, but in recent commits I've changed the build system to Cmake, I won't update the wiki until at least a few weeks before the final release, hopefully before the end of this year, **hopefully**.

>Even when I tried kdevelop, "open project" doesn't see any files. I see a .kdev4 file in there but kdevelop open project shows nothing, even though it says "all files" on the bottom. "about kedevelop" says 5.6.2. Do we need to use an old version of kdevelop or something?

I'm using Arch and it's working fine for me, just accepting all defaults. Maybe is a bug in the version you are using, not sure I don't use it for regular development, else try using Qt Creator, or VSCode.

bkw777, 2021-07-22:
Tried qtcreator, no dice there either, different problem though in that case.

Isn't there "reference" directions?

"use any ide you want" doesn't mean anything and it doesn't say how to build. I will happily configure the build steps into whatever ide I want. What are they?

bkw777, 2021-07-22:
Thank you for the directions in the other issue:
```
cd webcamoid
cmake -S .  -B ../webcamoid-build
make -C ../webcamoid-build
```

closed 2022-02-15 by hipersayanX