webcamoid/webcamoid #440: [Idea and Feature request] webcamoid & gscan2pdf

Wikinaut, 2021-06-14:
Hi, I just bought CZUR SHINE Scanner (Webcam) and start using it. On Linux, only a few programs are working with (cheese does not, but fswebcam, gstreamer do) and webcamoid works also, bravo.

What I found missing is a way to make multiple scans like the program gscan2pdf -- which unfortunately does not detect the CZUR SHINE (ET13R) device.

So, just as an idea, – but I do not know, how this can be achieved, – I just wrote down my wish to have some of the functionalities (webcamoid and gscan2pdf) combined. *)

Please feel free to close this issue if you think, that my idea is too far away from your original intention/s for webcamoid - I like that program and will send you a donation later.

*) perhaps I should contact the gscan2pdf maintainer, too, in order to fix the problem, why CZUR SHINE is detected by webcamoid but not by gscan2pdf - perhaps, because this only uses sane-devices, not uvc.

hipersayanX, 2021-06-14:
As long as it's just capturing images from a scanner, it could be possible, but saving as a pdf make no sense. The point of Webcamoid is capturing images and videos, apply effects, and save or stream the result.

Wikinaut, 2021-06-14:
I don't wanted primarily PDF, but the images, of course!

Wikinaut, 2021-06-14:
So, basically, there seems to be no easy way of a "uvc-to-sane" or "sane-to-uvc" let's call it driver or gateway.

Perhaps you can add to webcamoid a certain "batch" mode, so that shot images are shown as small preview images; I have no concrete idea at the moment.

Thanks for your swift answer.

hipersayanX, 2021-06-14:
I was watching some videos, and it seems that the scanner is basically a high resolution and low fps webcam, and is detected like that by the operating system, and the scanner is sending the frames either as a H264 or MJPG stream, that's the reason because Webcamoid can detect it :smile:

https://www.youtube.com/watch?v=MkaBPon9T0E

>So, basically, there seems to be no easy way of a "uvc-to-sane" or "sane-to-uvc" let's call it driver or gateway.

Actual scanners works in a different way compared to the one you have. Scanners try to keep scanning light smooth throughout the document when they capture the image, that's why scanners has a cover and a slow moving light rail, in the video above if you look closely,  the resulting captures always has a [light spot](https://youtu.be/MkaBPon9T0E?t=300) coming from either the scanner or ambient light.
Real scanners needs a completely different type of driver.

Wikinaut, 2021-06-14:
> Real scanners needs a completely different type of driver.

Yes, I do know this, but it would be a "nice to have" anyway.

closed 2021-06-14 by Wikinaut