webcamoid/webcamoid #441: Distorted video stream on macOS [webcamoid v8.8.0]

ashwinnaircy, 2021-07-14:
We are testing our UVC compatible camera with webcamoid v8.8.0 on macOS Big Sur 11.2.3. One of the resolutions that the camera supports is 2052x1536@30fps. This resolution is streaming correctly in: 
- Webcamoid 8.8.0 on Windows 10 
- QuickTime player on macOS Big Sur

However, when this resolution is tested in Webcamoid v8.8.0 on macOS Big Sur 11.2.3, we get the following distorted stream
[image]. 

I have verified with a USB trace analyzer that there is no error in the data being sent from the device to the Host. This is also confirmed by the same resolution streaming correctly on QuickTime player. 

I would like to know if : 

1. This is a known issue, or limitation from the application side
2. There is some way to get additional logs/traces from the application that will be useful for this debug

Thanks
Ashwin

hipersayanX, 2021-07-15:
>This is a known issue, or limitation from the application side

Nope, you are the first one reporting this issue.

>There is some way to get additional logs/traces from the application that will be useful for this debug

Nope, Webcamoid has a very rudimentary logs. isn't possible to debug such error unless you dig in the source code.

For me it's weird that this is working fine in Windows but not in Mac, since the code for decoding and scaling the frames is the same in both platforms.
You are getting the frames like that because the frame width isn't multiple of 32, so it it's causing an alignment problem. Does your camera supports other resolutions? In Windows it also informs the same resolution?

ashwinnaircy, 2021-07-15:
Thanks for the response.

> Does your camera supports other resolutions?
Yes, my camera does support other resolutions and all other are working fine on macOS + webcamoid too. Other resolutions that I have are: 
1. 1920x1080
2. 3840x2160
3. 4160x3120

> In Windows it also informs the same resolution?
Yes, it does. Please find below the capture from USBView showing the frame descriptor
[image]

> You are getting the frames like that because the frame width isn't multiple of 32, so it it's causing an alignment problem. 

This seems correct because all other resolutions are divisible by 32. Is this a macOS-requirement or a webcamoid-specific requirement?

hipersayanX, 2021-07-15:
>Is this a macOS-requirement or a webcamoid-specific requirement?

Is more like vendors doing thing they should not do. As as said before, is a bit alignment bug, I'll fix it later when I've some time.

hipersayanX, 2022-02-15:
Problem solved in 9.0.0, closing.

closed 2022-02-15 by hipersayanX