webcamoid/webcamoid #442: No Live preview or Camera Feed in WINPE 10

shadowwalkersteam, 2021-07-15:
I'm trying to use Webcamoid in WINPE but its not working there. I've installed all the camera drivers etc and everything is fine in WinPE. 
I tested my webcam on webcamtests.com in WINPE environment and there the Webcam was working perfectly but not in webcamoid. What could be the issue? Can anyone helo me in this issue?

hipersayanX, 2021-07-15:
Is the webcam detected or not at all? Vendor and model of the camera?

shadowwalkersteam, 2021-07-16:
Yes Webcam detected. It is USB HD WEBCAM.

hipersayanX, 2021-07-16:
>Yes Webcam detected.

Can you show me an screenshot of this part?

[image]

>It is USB HD WEBCAM (that is what is showing in device manager under Cameras)

Is a generic or integrated webcam? because I need to know which native formats are supported by that camera, otherwise i can't do anything with the information you are giving me.

shadowwalkersteam, 2021-07-20:
@hipersayanX  
[image]
[image]

It is integrated webcam.

hipersayanX, 2021-07-20:
Everything seems to be ok in the capture, the webcam and all it's parameters are detected, it an obvious question but, it could be that you forgot to click the Start/Stop button maybe?

shadowwalkersteam, 2021-07-26:
@hipersayanX I tried that as well but no success. I clicked Start/Stop buttons multiple times.

shadowwalkersteam, 2021-07-26:
@hipersayanX I've tried this application https://www.sordum.org/8585/webcam-on-off-v1-4/
and it is also working fine.

hipersayanX, 2021-07-27:
Did you tried the [daily build](https://github.com/webcamoid/webcamoid/releases/tag/daily-build), does it have the same problem?

hipersayanX, 2022-02-15:
Problem solved in 9.0.0, closing.

closed 2022-02-15 by hipersayanX