webcamoid/webcamoid #447: Webcamoid / no sound on play back (cheese has sound ok)

ruirai, 2021-08-03:
Webcamoid  Version        : 8.8.0-2.1
Operating System: openSUSE Tumbleweed 20210801 Qt Version: 5.15.2 Kernel Version: 5.13.4-1-default (64-bit)

I have USB camera Logitech C525 connceted to Intel NUC8i7BEH USB wire.
Creative T40 loud speakers are connect to NUC with 3.5 mm plug.

When I record something with C525 every thing seem to fine and "led"  flashes according to the sound level.
But when I play back my video (recoded with C525) I cannot hear sound at all (picture is fine).

When I use Cheese application for recording the play back has both sound and picture.

I will be happy to send you more information (attachments) because my language skills are not very good.

[inxi-F.txt](https://github.com/webcamoid/webcamoid/files/6926761/inxi-F.txt)
[webcamoid.log](https://github.com/webcamoid/webcamoid/files/6926763/webcamoid.log)
[pacmd.txt](https://github.com/webcamoid/webcamoid/files/6926839/pacmd.txt)

hipersayanX, 2021-08-04:
Have you selected the right input? for example, if you want to record from your microphone, you must select your microphone, if you are playing a video you must select the video as source.

[image]

ruirai, 2021-08-04:
I tried both Logitech C525 and Logitech C525 Mono. No sound on play back.
[image]

hipersayanX, 2021-08-05:
This happens with any recording format?

>Is there any list or logs I can get for you?

For the moment Webcamoid logs are barely useful, that is a feature I must work on when I have some time.

ruirai, 2021-08-05:
Yes, I have tried several recording format but result is the same -no sound on play back. I can reproduce the problem with my mother's openSUSE Leap 15.3  which has Webcamoid version 8.8.0-bp153.1.2 (I have openSUSE Tumbleweed with version Webcamoid 8.8.0-2.1).

ruirai, 2021-08-08:
I will write a bug report on openSUSE LEAP-15.3.

closed 2021-08-08 by ruirai