webcamoid/webcamoid #451: Crash on startup - MacOS OS X 10.14.6 Mojave.

FTL1, 2021-08-25:
# Reporting a problem?
Yes
## Summary
New and or fresh install does not open/run

Write here a brief description of the problem.
After downloading the latest version of webcamoid (8.8.0) I attempt to open/run the app.. and it immediately crashes ..and then the apple crash reporter pops up (see it's contents, below).

## Current Behavior
Webcamoid crashes on attempt to open.

## Expected Behavior
Webcamoid opens and runs(?)

How it should have work?
Dunno.. it has never successfully opened and ran.

## Steps to Reproduce
Download either version of the mac 8.8.0 dmg (same problem with either build)
Open
Drag and drop into apps
Run 
Allow run (if macos warns on "not from frigging applestore ish.."
Walk through install wizard (if this runs)
CRASH ... report to apple.

How would you solve the problem?
Figure out which other daemons/apps/plugins are contending for the camera (or other?) resources.

## Your Environment

* Webcamoid information: (version) (architecture)
  - 8.8.0
* Operating System information: (name) (numeric version and codename if applied) (architecture)
  - macos OS X 10.14.6 Mojave on a 2017 imac Retina 5K, 27 inch, with 4.2Ghz core i7 7700K, 64GB 2400Mhz DDR4, Radeon Pro 570 4GB
* Any other useful information: (logs, gdb backtrace, valgrind logs, screenshots, hardware, etc.)
  I have OBS installed and it has a virtual camera plugin.  
  I have Techsmith's SnagIt running, it has a plugin for screen video capture (this plugin is not in the directory where the camera plugins are).

Process:               webcamoid [1902]
Path:                  /Applications/webcamoid.app/Contents/MacOS/webcamoid
Identifier:            com.webcamoidprj.webcamoid
Version:               0
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           webcamoid [1902]
User ID:               501

Date/Time:             2021-08-25 11:23:13.796 -0400
OS Version:            Mac OS X 10.14.6 (18G9216)
Report Version:        12

Time Awake Since Boot: 60000 seconds

System Integrity Protection: disabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Application Specific Information:
dyld2 mode
abort() called

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff7118b2c2 __pthread_kill + 10
1   libsystem_pthread.dylib       	0x00007fff71246bf1 pthread_kill + 284
2   libsystem_c.dylib             	0x00007fff710f56a6 abort + 127
3   org.qt-project.QtCore         	0x00000001038a9959 0x103891000 + 100697
4   org.qt-project.QtCore         	0x0000000103bea8ee QMessageLogger::fatal(char const*, ...) const + 196
5   org.qt-project.QtGui          	0x000000010311cbad 0x102c94000 + 4754349
6   org.qt-project.QtGui          	0x0000000102ccce6d QGuiApplicationPrivate::createPlatformIntegration() + 7085
7   org.qt-project.QtGui          	0x0000000102ccce8b QGuiApplicationPrivate::createEventDispatcher() + 27
8   org.qt-project.QtCore         	0x0000000103a82efc QCoreApplicationPrivate::init() + 1612
9   org.qt-project.QtGui          	0x0000000102cc74e9 QGuiApplicationPrivate::init() + 57
10  org.qt-project.QtWidgets      	0x00000001026eb86a QApplicationPrivate::init() + 26
11  com.webcamoidprj.webcamoid    	0x0000000101fbf0ac 0x101fb9000 + 24748
12  libdyld.dylib                 	0x00007fff710503d5 start + 1

Thread 1:
0   libsystem_pthread.dylib       	0x00007fff712433f0 start_wqthread + 0

Thread 2:
0   libsystem_pthread.dylib       	0x00007fff712433f0 start_wqthread + 0

Thread 3:
0   libsystem_pthread.dylib       	0x00007fff712433f0 start_wqthread + 0

Thread 0 crashed with X86 Thread State (64-bit):
  rax: 0x0000000000000000  rbx: 0x00000001064a35c0  rcx: 0x00007ffeedc46678  rdx: 0x0000000000000000
  rdi: 0x0000000000000307  rsi: 0x0000000000000006  rbp: 0x00007ffeedc466b0  rsp: 0x00007ffeedc46678
   r8: 0x00000000000130a8   r9: 0x00007fffa77b9048  r10: 0x0000000000000000  r11: 0x0000000000000206
  r12: 0x0000000000000307  r13: 0x00007ffeedc468d8  r14: 0x0000000000000006  r15: 0x000000000000002d
  rip: 0x00007fff7118b2c2  rfl: 0x0000000000000206  cr2: 0x00007fffa77b7188
  
Logical CPU:     0
Error Code:      0x02000148
Trap Number:     133

Binary Images:
       0x101fb9000 -        0x1020e0ff7 +com.webcamoidprj.webcamoid (0) <58F97C1A-0EC7-399C-AAA6-1A0A42A44372> /Applications/webcamoid.app/Contents/MacOS/webcamoid
       0x1020fd000 -        0x10216cff3 +libavkys.8.8.0.dylib (0) <2F20183A-80DB-3933-AF67-9C241ADC9D85> /Applications/webcamoid.app/Contents/Frameworks/libavkys.8.8.0.dylib
       0x1021af000 -        0x1024f4ff3 +org.qt-project.QtQuick (5.15 - 5.15.2) <352AB469-1DE8-35E9-B278-2587D13DD56E> 
<<<======lots of other bins clipped=====>>>

External Modification Summary:
  Calls made by other processes targeting this process:
    task_for_pid: 0
    thread_create: 0
    thread_set_state: 0
  Calls made by this process:
    task_for_pid: 0
    thread_create: 0
    thread_set_state: 0
  Calls made by all processes on this machine:
    task_for_pid: 13392438
    thread_create: 0
    thread_set_state: 0

VM Region Summary:
ReadOnly portion of Libraries: Total=396.9M resident=0K(0%) swapped_out_or_unallocated=396.9M(100%)
Writable regions: Total=814.1M written=0K(0%) resident=0K(0%) swapped_out=0K(0%) unallocated=814.1M(100%)
 
                                VIRTUAL   REGION 
REGION TYPE                        SIZE    COUNT (non-coalesced) 
===========                     =======  ======= 
Activity Tracing                   256K        1 
Kernel Alloc Once                    8K        1 
MALLOC                           196.1M       16 
MALLOC guard page                   24K        4 
MALLOC_MEDIUM (reserved)         224.0M        2         reserved VM address space (unallocated)
MALLOC_NANO (reserved)           384.0M        1         reserved VM address space (unallocated)
STACK GUARD                       56.0M        4 
Stack                             9752K        4 
__DATA                            23.9M      242 
__FONT_DATA                          4K        1 
__LINKEDIT                       227.2M       13 
__TEXT                           169.7M      240 
__UNICODE                          564K        1 
mapped file                       25.9M        2 
shared memory                       44K        5 
===========                     =======  ======= 
TOTAL                              1.3G      537 
TOTAL, minus reserved VM space   709.2M      537 

Model: iMac18,3, BootROM 429.120.4.0.0, 4 processors, Intel Core i7, 4.2 GHz, 64 GB, SMC 2.41f1
Graphics: kHW_AMDRadeonPro570Item, Radeon Pro 570, spdisplays_pcie_device, 4 GB
Memory Module: BANK 0/DIMM0, 16 GB, DDR4, 2400 MHz, 0x859B, 0x435431364734534644383236362E4D313646450
Memory Module: BANK 0/DIMM1, 16 GB, DDR4, 2400 MHz, 0x859B, 0x435431364734534644383234412E433136464844
Memory Module: BANK 1/DIMM0, 16 GB, DDR4, 2400 MHz, 0x859B, 0x435431364734534644383236362E4D313646450
Memory Module: BANK 1/DIMM1, 16 GB, DDR4, 2400 MHz, 0x859B, 0x435431364734534644383234412E433136464844
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x16F), Broadcom BCM43xx 1.0 (7.77.61.3 AirPortDriverBrcmNIC-1305.12)
Bluetooth: Version 6.0.14d12, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en1
Serial ATA Device: Micron_1100_MTFDDAK2T0TBN, 2.05 TB
USB Device: USB 3.0 Bus
USB Device: USB3.0 Hub
USB Device: Bluetooth USB Host Controller
USB Device: FaceTime HD Camera (Built-in)
USB Device: ADB2USB
USB Device: USB2.0 Hub
USB Device: Sound Blaster Play! 3
USB Device: Magic Trackpad 2
USB Device: Sound Blaster Play! 3
USB Device: External
Thunderbolt Bus: iMac, Apple Inc., 41.1

slhck, 2021-09-27:
Crash here as well on macOS 11.5.2 with webcamoid 8.8.0.

Details:

```
Process:               webcamoid [36526]
Path:                  /Users/USER/*/webcamoid.app/Contents/MacOS/webcamoid
Identifier:            com.webcamoidprj.webcamoid
Version:               0
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
User ID:               501

Date/Time:             2021-09-27 13:49:47.305 +0200
OS Version:            macOS 11.5.2 (20G95)
Report Version:        12
Bridge OS Version:     5.5 (18P4759a)
Anonymous UUID:        D92D65EC-3952-33EB-A001-6C9D8D2F2C6A

Sleep/Wake UUID:       75D20956-637C-4ABA-9B77-603C9572E243

Time Awake Since Boot: 680000 seconds
Time Since Wake:       19000 seconds

System Integrity Protection: enabled

Crashed Thread:        9  Dispatch queue: com.apple.root.default-qos

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    Namespace TCC, Code 0x0

Thread 0:: Dispatch queue: com.apple.main-thread
0   com.apple.CoreFoundation      	0x00007fff2040cb1d _CFArrayCheckAndGetValueAtIndex + 119
1   com.apple.CoreFoundation      	0x00007fff2040ca51 -[__NSCFArray objectAtIndex:] + 74
2   com.apple.CoreFoundation      	0x00007fff20428933 CFArrayGetFirstIndexOfValue + 181
3   com.apple.CoreFoundation      	0x00007fff20417964 __CFRepositionTimerInMode + 70
4   com.apple.CoreFoundation      	0x00007fff20489f42 CFRunLoopTimerSetNextFireDate + 494
5   libqcocoa.dylib               	0x0000000107837bbb 0x107800000 + 228283
6   libqcocoa.dylib               	0x0000000107838704 0x107800000 + 231172
7   libavfoundation.dylib         	0x000000010e14d4c7 CaptureAvFoundation::updateDevices() + 167
8   libavfoundation.dylib         	0x000000010e14d3dd CaptureAvFoundation::CaptureAvFoundation(QObject*) + 237
9   libavfoundation.dylib         	0x000000010e146fca non-virtual thunk to Plugin::create(QString const&, QString const&) + 74
10  libavkys.8.8.0.dylib          	0x0000000101846890 AkElement::loadSubModule(QString const&, QString const&) + 400
11  libVideoCapture.dylib         	0x000000010e124410 VideoCaptureElementPrivate::captureLibUpdated(QString const&) + 112
12  libVideoCapture.dylib         	0x000000010e124135 VideoCaptureElement::VideoCaptureElement() + 373
13  libVideoCapture.dylib         	0x000000010e123533 VideoCapture::create(QString const&, QString const&) + 115
14  libavkys.8.8.0.dylib          	0x0000000101844a21 AkElement::createPtr(QString const&, QString const&) + 513
15  libavkys.8.8.0.dylib          	0x00000001018447b5 AkElement::create(QString const&, QString const&) + 21
16  com.webcamoidprj.webcamoid    	0x00000001017203b5 0x1016f2000 + 189365
17  com.webcamoidprj.webcamoid    	0x00000001017168f1 0x1016f2000 + 149745
18  com.webcamoidprj.webcamoid    	0x00000001016f8d7c 0x1016f2000 + 28028
19  com.webcamoidprj.webcamoid    	0x00000001016f8777 0x1016f2000 + 26487
20  libdyld.dylib                 	0x00007fff2036ff3d start + 1

Thread 1:
0   libsystem_pthread.dylib       	0x00007fff20350420 start_wqthread + 0

Thread 2:
0   libsystem_pthread.dylib       	0x00007fff20350420 start_wqthread + 0

Thread 3:
0   libsystem_pthread.dylib       	0x00007fff20350420 start_wqthread + 0

Thread 4:
0   libsystem_pthread.dylib       	0x00007fff20350420 start_wqthread + 0

Thread 5:: Thread (pooled)
[... 570 lines cut ...]
Memory Module: BANK 2/ChannelB-DIMM0, 4 GB, LPDDR3, 2133 MHz, Samsung, K4E6E304EC-EGCG
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x7BF), wl0: Jul  7 2021 02:34:10 version 9.30.444.17.32.5.70 FWID 01-4075fabc
Bluetooth: Version 8.0.5d7, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: Extreme SSD
USB Device: USB3.0 Hub
USB Device: USB3.0 Hub
USB Device: Logitech StreamCam
USB Device: USB 3.1 Bus
USB Device: USB2.0 Hub
USB Device: USB2.0 Hub
USB Device: eLicenser
USB Device: USB-C Digital AV Multiport Adapter
USB Device: BillBoard Device
USB Device: Apple T2 Bus
USB Device: Touch Bar Backlight
USB Device: Touch Bar Display
USB Device: Apple Internal Keyboard / Trackpad
USB Device: Headset
USB Device: Ambient Light Sensor
USB Device: FaceTime HD Camera (Built-in)
USB Device: Apple T2 Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 47.5
Thunderbolt Bus: MacBook Pro, Apple Inc., 47.5
```

Leemusa, 2021-12-24:
Solution. I had the same issue on my MacOS Mojave.

Somehow I crashed Webcamoid by selecting the wrong virtual camera - I don't know, it happened so fast and I was never able to start the program again to see exactly which camera I had selected.

The solution was to follow the directions made by  j256  in this post:

https://github.com/webcamoid/webcamoid/issues/144

Some how this issue was closed but the problem was never solved: once a broken virtual camera is selected Webcamoid will enter into a crash loop and never recover.

The solution to this is rather easy: during Webcamoid start check to see if the app crashed the last time it exited. If it did start the app with a safe camera selection that is guaranteed not to crash the app.

hipersayanX, 2021-12-27:
@Leemusa try installing the daily build, it should be be fixed already there.

>The solution to this is rather easy: during Webcamoid start check to see if the app crashed the last time it exited. If it did start the app with a safe camera selection that is guaranteed not to crash the app.

Open an issue for that, but don't expect that feature to be implemented in the not even near time.

slhck, 2021-12-29:
The application now starts up again. So this issue is fixed in the daily build.

That said, I do not see any camera preview in the app. It just shows this screen:

[image]

The camera's green light is on.

hipersayanX, 2021-12-29:
@slhck click in the icon in the top right corner, Video, click in the camera to to select it, and once again to enter to its settings, which formats does it shows?

[image]

slhck, 2021-12-29:
[image]

I see no settings below like what you have. Perhaps this is worth opening another issue for.

I should add that none of my webcams works with this program. I have a Logitech StreamCam connected over USB-C that also shows no output in Webcamoid.

hipersayanX, 2021-12-29:
>Video format: v024

That's the problem, that format isn't supported, also not sure which format is that. Does your camera provides other formats? maybe YUV, MJPEG, H264, or similar?

>I see no settings below like what you have. Perhaps this is worth opening another issue for.

Nope, the reason because you haven't the camera controls is because AVFoundation doesn't provide an API for that, isn't a fixable thing.

>I should add that none of my webcams works with this program. I have a Logitech StreamCam connected over USB-C that also shows no output in Webcamoid.

Can you tell me which formats are reported by those cameras?

slhck, 2021-12-29:
So as to allow you to close this thread once a new release is made (since the bug is fixed), I created a new issue: https://github.com/webcamoid/webcamoid/issues/477

hipersayanX, 2022-02-15:
Problem solved in 9.0.0, closing.

closed 2022-02-15 by hipersayanX