webcamoid/webcamoid #455: website shows webcamoid picture even though i cant find it's drivers or the app

joku0, 2021-09-07:
I had webcamoid at some point but i deleted it. Now this one website shows this image: 
[image]
while other websites show my obs virtual camera as i want them to. 
I tried following another thread with a similar problem where i would delete the driver from regedit, but it only showed my obs virtual camera there.
How do i delete webcamoid completelly off my pc?

hipersayanX, 2021-09-08:
It's all explained in #178, look [here](https://github.com/webcamoid/webcamoid/issues/178#issuecomment-606301900) and [here](https://github.com/webcamoid/webcamoid/issues/178#issuecomment-614857556). Also, I forgot to mention there but you must also delete the virtual camera service, opening an administrator CMD and running:

```
sc delete AkVCamAssistant
```

This gonna be fixed in Webcamoid 9.0.0 for sure.

joku0, 2021-09-08:
This worked, thanks!

closed 2021-09-08 by joku0