webcamoid/webcamoid #459: After using update script on 32-bit system, program opens infinite new windows on launch in quick succession

Gradivis, 2021-10-02:
Reporting a Problem:
## Summary
I am using Twister OS on Raspberry Pi - it's a 32-bit OS. I installed webcamoid from the linux built-in software repository and it worked great. I clicked on the notification that there was an update available, and downloaded the linux update not realising it was for 64-bit. I ran the package through terminal, and on next launch of webcamoid it behaves strangely. It will perpetually open new instances/windows of itself, each getting slightly smaller. They open in quick succession and keep opening until I close the application. I have uninstalled and reinstalled, but the behaviour remains. Totally my fault for using a 64-bit update on my 32-bit system - it was early in the morning! Any help you can offer would be great, I love your app!!

## Current Behavior
The program will launch an infinite number of instances/windows of itself in quick succession (like the animation from Solitaire when you win the game). It will keep crashing in this way until manually quit.

## Expected Behavior
Open a single window as before.

## Steps to Reproduce

1. Install Webcamoid via Built-In Software Repository
2. Download and run 64-bit update package from github
3. Launch webcamoid

## Suggestions and tips

I would remove any and all files related to webcamoid and start from fresh. The problem is I am not sure where any modifications the update script I downloaded are being made.

## Your Environment

* Latest Version - 8.8.0
* Operating System information: Twister OS 2.1.2 for Raspberry Pi (32-bit) - Based on Raspbian

Thanks!

hipersayanX, 2021-10-02:
This

>I am using Twister OS on **Raspberry Pi** - it's a **32-bit OS**.

and this

>Totally my fault for using a **64-bit update** on my 32-bit system

isn't technically possible, those pre-compiled binaries only works on AMD64 compatible CPUs, while Raspberrys (at least latest models) comes with an ARM64 CPU, AMD64 won't work in an ARM64, if you try to launch the program, it will just throw an error related to wrong architecture. The same happens if you are using a 32 system it will just never work or run any code.

I will tell you what probably happened: you probably tried to launch the installer or the AppImage, either from terminal, ether from GUI,  it throwed an error (probably silently) which you ignored, and then you proceeded to launch Webcamoid as normal from the app menu, which means you were executing the version installed from your distro's repositories.

>They open in quick succession and keep opening until I close the application. I have uninstalled and reinstalled, but the behaviour remains.

Do you mean something like this? with each windows inside each other?

[image]

I won't tell you how to solve this "bug", but I will tell you this: try making Webcamoid's window smaller and moving it around, and try to figure what is happening :smile: 

>Operating System information: Twister OS 2.1.2 for Raspberry Pi (32-bit) - Based on **Raspbian**

Debian keeps distributing a 2 years old, deprecated, and full of bugs, version, and no one seems to be maintaining it.
Please, if some Debian user is reading this: **tell Debian developers to update the package to the latest version.**

Meanwhile, you can use a more up to date distro like [Arch Linux ARM](https://archlinuxarm.org/), I personally maintain the AUR packages.

Gradivis, 2021-10-03:
Thanks so much for your reply - I suspect you are correct in your analysis of what happened. I didn't see any errors, but I also didn't see any confirmation messages and decided to launch webcamoid to see if anything had changed.

I took your advice and tried to play around with the window size and positioning - [here's the result](https://github.com/Gradivis/Pira/blob/8d4f58cb1982ed972c8bdf3137411d79c28611f5/ezgif.com-gif-maker%202.gif). 

Strangely, I did a complete re-install with a fresh copy of the OS on a new SD card, and then installed from the repository as I did before. The same issue is happening from first launch, so (1) I am quite confused as to how it was working the very first time I launched, (2) leaning towards this being caused by the package in the repo itself?

I'm more attached to using webcamoid which (for the first launch when it was working) seemed absolutely perfect for what I'm trying to do - it was outputting my USB microscope amazingly. So I'll give Arch ARM a try today and appreciate the suggestion. Been years since I used Arch!

Many thanks for your response once again! Should I close this issue seeing as it's likely to do with OS/package manager issues and not webcamoid? Cheers.

hipersayanX, 2021-10-03:
>I took your advice and tried to play around with the window size and positioning - here's the result.

And did you not figured it ut? it's capturing the desktop :smile:

>Strangely, I did a complete re-install with a fresh copy of the OS on a new SD card, and then installed from the repository as I did before. The same issue is happening from first launch, so (1) I am quite confused as to how it was working the very first time I launched,

Just select the Raspberry camera.
I don't know how did you partitioned your SD, but if you made one partition for the system and one for /home, and you reinstalled the system without touching the /home partition, all your configurations will be preserved, .

>(2) leaning towards this being caused by the package in the repo itself?

No, as said before, just select the Raspberry camera to capture from it.

>Should I close this issue seeing as it's likely to do with OS/package manager issues and not webcamoid?

Sure, if you think you are done with all questions, close it.

Tribble42, 2021-10-04:
Same thing happens to me .... infinite-vanishing-into-the-distance screens when I open the app.   Ubuntu 20.0.
[image]

Gradivis, 2021-10-05:
I have to admit I had a pretty good chuckle when I realised what capture source I was on...in my defense it seems my camera was no longer being recognised but I fixed that issue and can now select it as a source. All working great - thanks so much for putting me on the right track!

Gradivis, 2021-10-05:
> Same thing happens to me .... infinite-vanishing-into-the-distance screens when I open the app. Ubuntu 20.0. 
I had to reinstall my camera and then select it as a source. I don't see a camera on your list of devices there. For me once I reinstalled the camera drivers, it showed up in the list so this was definitely user error and I am closing the issue :)

closed 2021-10-05 by Gradivis