webcamoid/webcamoid #461: Feature request: visual audio preview

phhu, 2021-10-11:
Would it be possible to add a visual audio preview feature to this program?

Several times I have used it to record webcam video, only to find that the recording produced had no audio, probably because the microphone was muted. To make this less likely to happen, it would be useful to have a graphical preview of audio input: perhaps a simple audio level meter, which would move when sound is received. This would give an immediate indication that audio is being received (via the selected audio input), just as we have video preview on the main video record screen. 

(This would be particularly useful when multiple audio inputs are available, and it may not be immediately obvious which one is being used for recording)

hipersayanX, 2022-06-03:
@iam-TJ Maybe creating a special ItemDelegate with a vumeter on bottom and then replace the ItemDelegate for audio devices with that?

[image]

Anyway, audio inputs/outputs need some rework, I was thinking in using an approach similar to OBS, and allow mixing audio and video sources, instead of playing and recording individual sources.

iam-TJ, 2022-06-03:
Whilst I'm playing about with it would this be the correct place to 'steal' the samples from?
```
diff --git a/StandAlone/src/audiolayer.cpp b/StandAlone/src/audiolayer.cpp
index 137e32be..0623bd1e 100644
--- a/StandAlone/src/audiolayer.cpp
+++ b/StandAlone/src/audiolayer.cpp
@@ -686,6 +686,7 @@ AkPacket AudioLayer::iStream(const AkPacket &packet)
         && this->d->m_audioInput.contains(this->d->m_input)) {
         (*this->d->m_audioSwitch)(packet);
     }
+    // TODO: feed GUI visual audio level monitor from here?
 
     this->d->m_mutex.unlock();
```
In terms of the visual indicator I was thinking much more minimalist - it really doesn't need to take up much screen space. I'm currently playing with a single pixel horizontal (at top) or vertical (left side) using the full {width|height} as the scale from -200db (silence) to 0db (full volume).
Also, to avoid excessive processing, the samples would be averaged and the visual indicator only updated 10 or 20 times each second.
The monitor would attach to whatever the current input source device is.

hipersayanX, 2022-06-06:
Depends from where you want to take the audio samples. If you want to take the input from mics you must the the samples from audioIn, if you want to take the samples that will be recorded you must take it from audioSwitch.

[image]

phhu, 2022-06-07:
> I'm currently investigating this. Would a simple vertical level indicator be sufficient?

For my purposes, yes, just a vertical meter that moves as the audio level changes - so that excess level and so distortion can be detected as well as absence of input. (Probably the operating system is good enough to figure out where sound is / should be coming from where sound is missing).