webcamoid/webcamoid #468: Webcamoid app crashes on Win10 when selecting camera input

ingineru-de-sistem, 2021-11-09:
Hello

Running Webcamoid 8.8.0 using qt 5.15.2 on a Windows 10 Build 17763.rs5_release.180914-1434 .
It runs fine when capturing video outputted to any of the 2 monitors, but it crashes as soon as I switch to the camera. Have to disconnect the camera to be able to launch it again.
The camera is a Jelly Comb FHD camera using the standard Microsoft driver (works fine in Zoom).
USB camera data: VID=1BCF PID=28C4 REV=0530 MI=00 (whatever the last one means).
Should someone wish to repro with real hardware, here's a link https://www.amazon.ca/gp/product/B099NJNPVJ .

Please advise if you need any other data. (Or if it's a "code 18" and I need some other part installed that I missed...)

No rush (I just needed Webcamoid as a FOSS way to test my camera, but since I have to use Zoom and others for work, it's already been tested thoroughly and it works fine.)

Thanks

hipersayanX, 2021-11-11:
Do you have the same problem using the [daily build](https://github.com/webcamoid/webcamoid/releases/tag/daily-build)?

ingineru-de-sistem, 2021-11-12:
Hello
In the daily build (from 2021-11-11) it doesn't crash anymore when the camera is on.It may crash if the usb camera gets disconnected.But, most importantly:1) the menus are not being displayed, so yes, I can test my camera but little else.2) I cannot move the window to the secondary monitor (to avoid the tunnel effect when recording the same monitor).

Hope this helps.

hipersayanX, 2021-11-12:
>1) the menus are not being displayed, so yes, I can test my camera but little else

I don't understand this, which menus? can you post a screenshot?

ingineru-de-sistem, 2021-11-13:
The only two menu items in that build are "take a photo" and "record video".Nothing else.
Once I disconnect the camera, it switches to capturing screen 1 and that's it.
Pressing "take a photo" to get a screen shot freezes the app.Maybe that daily version was not a good vintage ?
See a screenshot pasted below (some icons have been removed for privacy).

Let me know when you have something more stable to test with.
Thanks.

hipersayanX, 2021-11-14:
>The only two menu items in that build are "take a photo" and "record video".Nothing else.

Don't you have the effects and the preferences buttons at the top?

[image]

>Pressing "take a photo" to get a screen shot freezes the app.Maybe that daily version was not a good vintage ?

I never experienced any bug like that, and I was using that functionality very recently.

>See a screenshot pasted below (some icons have been removed for privacy).

You did not posted any image, don't answer from your mail, answer directly in the issue thread and post the capture from there.

hipersayanX, 2022-02-15:
Problem solved in 9.0.0, closing.

closed 2022-02-15 by hipersayanX