webcamoid/webcamoid #469: Feature Request: M1 build for macOS

ajmas, 2021-11-18:
This feature request is simply to have support for ARM build targets, for macOS.

I did search through the issues and also running 'file' against the main binary in the package suggests the target is still only x86_64. It is possible that I missed something in my search.

Given that there is already support for Android and Raspberry Pi (based on [this](https://forums.raspberrypi.com/viewtopic.php?t=261027)), I would imagine the necessary adjustments would be minor?

hipersayanX, 2021-11-18:
I've no problem adding support for M1 chips, but afaik, [github environment](https://github.com/actions/virtual-environments/blob/main/images/macos/macos-11-Readme.md) runs on x86_64, and not sure how to cross compile to M1. Also, I'll be not able to test it because I haven't a M1 MacBook, so I'll keep supporting x86_64 until homebrew remove the support for it, and then, if nobody take the lead of Mac development, I'll just remove Mac support.

ajmas, 2021-11-18:
Actual user testing will depend on implicated humans being available, and we should take your comment as a request for volunteers. 

At the same time, it is possible to cross compile, using Intel CPUs. I looked into this for another project, but if I remember it was XCode based. I’ll link it here when I find it.

hipersayanX, 2021-11-18:
>and we should take your comment as a request for volunteers.

Take it and multiply by 1e6.

gandergo, 2022-12-01:
> [github environment](https://github.com/actions/virtual-environments/blob/main/images/macos/macos-11-Readme.md) runs on x86_64, and not sure how to cross compile to M1

Saw that the project has a [.cirrus.yml](https://github.com/webcamoid/webcamoid/blob/master/.cirrus.yml) in the root. Does this project currently uses Cirrus CI? They now offering [m1 mac](https://github.com/actions/runner/issues/805#issuecomment-769712066) build environment for public projects.

I tried to modify the existing .cirrus.yml for m1 build environment and modify for current build.sh for m1 in my [fork](https://github.com/gandergo/webcamoid/tree/mac-arm):
1. Add necessary [flags](https://github.com/gandergo/webcamoid/blob/e107a21e0a19f3d561d3bd98b96176ea103dc78a/ports/ci/mac-arm/build.sh#L29) for targeting m1 for Qt5:
   ```
   -DCMAKE_OSX_ARCHITECTURES=arm64
   export CFLAGS="-arch arm64 -target arm64-apple-macos11"
   export CXXFLAGS="-arch arm64 -mmacosx-version-min=11.0 -std=c++11"
   ```
2. I did not manage to properly sign the binary in CI and m1 mac would not allow the app to start without it. To make it work, similar to [here](https://github.com/supercollider/supercollider/issues/5603), I self-signed the binary locally.

Then program can then start with the akvirtualcamera plugin compiled on the m1 too, but i did not test many other features.

If such approach is considered acceptable, I can see if i can polish the method further and submit a pull request. Thanks.

hipersayanX, 2022-12-05:
@gandergo So basically no code changes were required?

>If such approach is considered acceptable, I can see if i can polish the method further and submit a pull request. Thanks.

This changes won't affect the build in FreeBSD, right? because I remember Cirrus was very limited for free builds.
Also looking at the link here

>I did not manage to properly sign the binary in CI and m1 mac would not allow the app to start without it. To make it work, similar to [here](https://github.com/supercollider/supercollider/issues/5603), I self-signed the binary locally.

Their are using the --deep flag, is this flag necessary? if so it can be added [here](https://github.com/webcamoid/DeployTools/blob/8a80f4206f8e02b0f8727e0941cd3ecba3f5cb66/WebcamoidDeployTools/DTMac.py#L316).

Also , thanks!

gandergo, 2022-12-18:
@hipersayanX sorry for my late reply. Have been away last week.

> So basically no code changes were required?

Yes, I only changed those build configurations. But I only tried some features like using it with the M1 build of virtual camera.

> This changes won't affect the build in FreeBSD, right? 

The Mac M1 build can be defined as a separate build task like [here](https://github.com/gandergo/webcamoid/blob/e107a21e0a19f3d561d3bd98b96176ea103dc78a/.cirrus.yml#L1). From the Cirrus CI doc [here](https://cirrus-ci.org/faq/#are-there-any-limits), it should not affect the existing limits for the builds of FreeBSD.

> Their are using the --deep flag, is this flag necessary? if so it can be added [here](https://github.com/webcamoid/DeployTools/blob/8a80f4206f8e02b0f8727e0941cd3ecba3f5cb66/WebcamoidDeployTools/DTMac.py#L316).

Yes, I think this flag is necessary. I did try adding this flag during the build as in [here](https://github.com/gandergo/DeployTools/blob/72c7d94dc0c84214eb00b1cf368047dda83f4abe/WebcamoidDeployTools/DTMac.py#L377), but it does not work and Mac on M1 still refuses to open the binary, unless I self-sign the binary again locally after the build. I still have not found out the cause on why this local re-signing is necessary.

hipersayanX, 2022-12-20:
> Yes, I think this flag is necessary. I did try adding this flag during the build as in [...]

It is what it is :man_shrugging: , to make a binary work for other users, the binary must be code signed, and for code signing you must register with an Apple ID and pay for the service, me as many other FOSS developers won't waste their time and money on that, most FOSS developers don't  make enough money to care about such things.
Just do a basic code sign, and make easy for the user to build and install with homebrew, and that's all, you don't need to worry about anything else.

rcoenen, 2023-03-17:
So what is the latest - can we use this on M1 Macs yes or no?

ajmas, 2023-03-17:
At this point having someone try building it on an M* Mac would be good and then identify any issues. @rcoenen if you are able to try building and documenting any build or run issues, I think that would be useful starting point.

Also there are now M1 runners and Pi compatible ARM runners:

 - https://github.com/marketplace/actions/arm-runner
 - https://github.com/github/roadmap/issues/528

Contributions to fix this up would be useful.

Myself, I don't have access to an ARM based Mac right now, to test locally.

hipersayanX, 2023-03-19:
I was about to write an answer here but at the end I decided to create an issue #584.

@ajmas 

>Myself, I don't have access to an ARM based Mac right now, to test locally.

Me too and don't expecting to have one in the short time, not even in long time.

hipersayanX, 2023-04-30:
Well, I've tried but

- The hosted system images provided by github are only x86_64.
- I can compile Webcamoid for ARM64, but it won't link with the libraries provided by homebrew because the libraries are x86_64 only.
- Apparently I can download install ARM64 packages with ```arch -arm64 brew install```, yet it won't work in x86_64.

So I reached a dead point with this issue. [Here](https://github.com/webcamoid/webcamoid/actions/runs/4846206713/jobs/8635605206) are the results.
Ideas welcome, otherwise we have no other option than waiting for github to add the ARM64 images.

ajmas, 2023-04-30:
I haven’t been able to contribute much recently due to day job.

hipersayanX, 2023-04-30:
I will try with Cirrus later.
@ajmas no problem.

hipersayanX, 2023-05-01:
And [there you have](https://github.com/webcamoid/webcamoid/releases/tag/daily-build), Webcamoid packages for the ARM based Macs, but can't test it so not sure if works or not.

boli, 2023-05-28:
> And [there you have](https://github.com/webcamoid/webcamoid/releases/tag/daily-build), Webcamoid packages for the ARM based Macs, but can't test it so not sure if works or not.

did not work for me on macbook air m1. i guess its a code signing thing:
```

Exception Type:        EXC_BAD_ACCESS (SIGKILL (Code Signature Invalid))
Exception Codes:       UNKNOWN_0x32 at 0x0000000102d94000
Exception Codes:       0x0000000000000032, 0x0000000102d94000

Termination Reason:    Namespace CODESIGNING, Code 2 Invalid Page
```

so i tried this
`
codesign --force --deep --sign - /Applications/Webcamoid.app`

...but it didnt help.

hipersayanX, 2023-05-29:
@boli maybe leaving it unsigned by default would help? the other option would be contacting with the guy maintaining the [homebrew formulae](https://formulae.brew.sh/cask/webcamoid) and tell him to compile straight from the sources instead of downloading it from the binary, that way users will be able to install Webcamoid from homebrew. I can't do nothing more beside that, I can't even test Webcamoid in Mac anymore, so now the development is on Mac community hands only.

hipersayanX, 2023-06-10:
The source code compile fine and should be able to work fine in Mac, but binary packages won't be provided anymore.

closed 2023-06-10 by hipersayanX

rokejulianlockhart, 2023-06-10:
> binary packages won't be provided anymore.

Why, @hipersayanX?

hipersayanX, 2023-06-10:
@rokejulianlockhart read #584, Webcamoid has $0 financial support, it's absolutely impossible for me to pay for code signing or buy a Mac, I had not any other option.

abibars, 2023-06-20:
good day gentlemen, i will be glad to support compile and sign the code, if you can  guide me through the the binary files

hipersayanX, 2023-06-20:
@abibars the documentation for code signing is [here](https://developer.apple.com/support/code-signing/) (I can't help you there because I never done that), the certificate can be bought in the sites listed [here](https://www.sslshopper.com/apple-code-signing-certificates.html).
If you want to compile for yourself, the documentation is [here](https://github.com/webcamoid/webcamoid/wiki/Build-and-install-in-Mac), also [here](https://github.com/webcamoid/webcamoid/blob/cd578a0545114ea871931e34a00a3854a7a604dd/ports/ci/mac/install_deps.sh#L31) you can find the dependencies.
If the idea is compiling and distributing officially for the project, is a matter of buying the certificate and then [adding it to the pipeline](https://github.com/webcamoid/webcamoid/actions/workflows/mac.yml), which I don't know how to do it yet.