webcamoid/webcamoid #470: Upgrade to Qt6

ajmas, 2021-11-18:
Would it be worth upgrading to Qt6? What work would need to be done and what help is needed?

Additionally, this issue is more of a place holder for to link issues that depend on Qt6, which at this point probably isn’t many?

hipersayanX, 2021-11-19:
>Would it be worth upgrading to Qt6?

Of course.

>What work would need to be done and what help is needed?

Its simply taking the code and upgrading it to Qt6, Webcamoid does not have more dependencies with Qt rater than QtCore, QtGui, QtQuick, QtQuickControls, and a few more from core modules, modules that are already ported to Qt6, porting is just fixing a few things here and there and that's all.
My plan was start porting to Qt6 soon after releasing Webcamoid 9 (there more plans for future releases but I will explain later). The only problem with porting to Qt6 is that Debian, and in consequence Ubuntu and it's derivatives, still [did not have packed Qt6 in their repositories](https://packages.debian.org/search?keywords=qt6&searchon=names&suite=stable&section=all), while [all other major distros already have it](https://pkgs.org/search/?q=qt6), but at this point, if Debian and derivatives don't care about maintaining their software updated, I don't care about Debian.

>Additionally, this issue is more of a place holder to link issues that depend on Qt6, which at this point probably isn’t many?

There is none, afaik.

ajmas, 2021-11-19:
I imagine initially this could exist as a branch, targeting what is supported, until Debian is ready?

I was looking into the CLI only build recipe last night, but didn’t get far.

hipersayanX, 2021-11-19:
>I imagine initially this could exist as a branch, targeting what is supported, until Debian is ready?

No way, I won't maintain two versions, I'll do a straight port and remove all Qt5 code, and Debian well... let see in a few 
years when when they decide to package Qt6 in their repositories.

iam-TJ, 2022-06-02:
I'm doing some initial research on what is required to make the codebase compatible with Qt6 based on [Porting from Qt 5 to Qt 6 using Qt5Compat library](https://www.qt.io/blog/porting-from-qt-5-to-qt-6-using-qt5compat-library). It looks like, if `qt5compat` is added to the build system there will be little or nothing to do. because the only affected `#include` statements do not include a path and therefore are usable as-is with `qt5compat`.
```
$  $ grep -s -rnE 'Q(LinkedList|RegExp|StringRef|XmlSimpleReader|TextCodec|TextEncoder|TextDecoder)' | grep '#include'
libAvKys/Plugins/VirtualCamera/src/v4l2lb/src/vcamv4l2lb.cpp:31:#include <QTextCodec>
libAvKys/Plugins/VirtualCamera/src/dshow/src/vcamdshow.cpp:32:#include <QTextCodec>
libAvKys/Plugins/VirtualCamera/src/akvcam/src/vcamak.cpp:32:#include <QTextCodec>
libAvKys/Plugins/VirtualCamera/src/cmio/src/vcamcmio.cpp:32:#include <QTextCodec>
libAvKys/Lib/src/akfrac.cpp:22:#include <QRegExp>
libAvKys/Lib/src/akpluginmanager.cpp:25:#include <QRegExp>
libAvKys/Lib/src/akplugininfo.cpp:24:#include <QRegExp>
```
With a simple additional to `CMakeList.txt` the project should build:
```
PUBLIC_LIBRARIES
Qt::Core5Compat
```
 I'll test this here on Debian. On Debian 11 Bullseye a simple count suggests almost all the same libraries are available:
```
$ apt list '*libqt5*' 2>/dev/null | wc -l; apt list '*libqt6*' 2>/dev/null | wc -l
139
131
```
Further comparison using a custom script to strip out the version number reports:
```
$ bin/debian-qt6-port-check
QT5 only = 19 ->  libqtcharts libqtcontacts libqtfeedback libqtgamepad libqtlocation libqtmultimediagsttools libqtorganizer libqtpublishsubscribe
 libqtscript libqtscripttools libqtserviceframework libqtsysteminfo libqttexttospeech libqtversit libqtversitorganizer libqtwebengine
 libqtwebkit libqtx11extras libqtxmlpatterns
QT6 only = 43 ->  libqtbodymovin libqtcore libqtcore5compat libqtlabsanimation libqtlabsfolderlistmodel libqtlabsqmlmodels libqtlabssettings libqtlabssharedimage
 libqtlabswavefrontmesh libqtopenglwidgets libqtpdfquick libqtqmlcore libqtqmllocalstorage libqtqmlxmllistmodel libqtquick3d libqtquick3dassetimport
 libqtquick3dassetutils libqtquick3deffects libqtquick3dhelpers libqtquick3diblbaker libqtquick3dparticles libqtquick3druntimerender libqtquick3dutils libqtquickcontrols2impl
 libqtquickdialogs2- libqtquickdialogs2quickimpl libqtquickdialogs2utils libqtquicklayouts libqtquicktimeline libqtremoteobjectsqml libqtscxmlqml libqtsensorsquick
 libqtshadertools libqtstatemachine libqtstatemachineqml libqtsvgwidgets libqtuitools libqtwaylandeglclienthwintegration libqtwaylandeglcompositorhwintegration libqtwebenginequick
 libqtwebenginequickdelegatesqml libqtwebviewquick libqtwlshellintegration
Common = 61 ->  libqt3danimation libqt3dcore libqt3dextras libqt3dinput libqt3dlogic libqt3dquick libqt3dquickanimation libqt3dquickextras
 libqt3dquickinput libqt3dquickrender libqt3dquickscene2d libqt3drender libqtbluetooth libqtconcurrent libqtdatavisualization libqtdbus
 libqtdesigner libqtdesignercomponents libqtgui libqthelp libqthunspellinputmethod libqtmultimedia libqtmultimediaquick libqtmultimediawidgets
 libqtnetwork libqtnetworkauth libqtnfc libqtopengl libqtpdf libqtpdfwidgets libqtpositioning libqtpositioningquick
 libqtprintsupport libqtqml libqtqmlmodels libqtqmlworkerscript libqtquick libqtquickcontrols2- libqtquickparticles libqtquickshapes
 libqtquicktemplates2- libqtquicktest libqtquickwidgets libqtremoteobjects libqtscxml libqtsensors libqtserialbus libqtserialport
 libqtsql libqtsvg libqttest libqtvirtualkeyboard libqtwaylandclient libqtwaylandcompositor libqtwebchannel libqtwebenginecore
 libqtwebenginewidgets libqtwebsockets libqtwebview libqtwidgets libqtxml
```

hipersayanX, 2022-06-03:
@iam-TJ [I did the port already](https://github.com/webcamoid/webcamoid/commits/Qt6) but there were a lot more problems than just Debian not having it in the repositories, it's all explained [here](https://github.com/webcamoid/webcamoid/commits/Qt6).

iam-TJ, 2022-06-03:
I totally missed the branch with there being no mention in the issue! I've just done a speed-read through the various bug reports and other resources and it looks like it's not as bad for Android as it looks!.

Firstly the two most relevant Qt bug reports:
[84382 QtAndroidExtras in Qt 6](https://bugreports.qt.io/browse/QTBUG-84382)
[90498 Permission handling API](https://bugreports.qt.io/browse/QTBUG-90498)

However, Qt6 **does** provide `RequestPermission()` via `#include <QtCore/private/qandroidextras_p.h>` and as the linked blog post says:

> Changes to Qt Android Extras
>
>Key functionality from the module has been brought over to other Qt modules.
>
>Clients that still rely on missing functionality can include the private header <QtCore/private/qandroidextras_p.h> as a stopgap solution.
>
>To enable private headers use QT += core-private with qmake, or add a project dependency to Qt::CorePrivate with CMake.

```
namespace QtAndroidPrivate
{ ...
#if QT_CONFIG(future)
    enum PermissionType {
        Camera,
        Microphone,
        Bluetooth,
        Location,
        PreciseLocation,
        BackgroundLocation,
        PreciseBackgroundLocation,
        BodySensors,
        PhysicalActivity,
        Contacts,
        Storage,
        Calendar
    };

    enum PermissionResult {
        Undetermined,qt6-base-private-dev
        Authorized,
        Denied
    };

    Q_CORE_EXPORT QFuture<QtAndroidPrivate::PermissionResult>
    requestPermission(QtAndroidPrivate::PermissionType permission);
    Q_CORE_EXPORT QFuture<QtAndroidPrivate::PermissionResult>
    requestPermission(const QString &permission);

    Q_CORE_EXPORT QFuture<QtAndroidPrivate::PermissionResult>
    checkPermission(QtAndroidPrivate::PermissionType permission);
    Q_CORE_EXPORT QFuture<QtAndroidPrivate::PermissionResult>
    checkPermission(const QString &permission);
#endif
...
```
That file is available in Debian Bullseye via package `qt6-base-private-dev` and I've the content above came from that file so cross-platform builds should be possible.

It may be worth revisiting this.