webcamoid/webcamoid #474: Photo Preview Icon On Lastest Daily Build Version Not Opening Captured Images/Videos.

shawnk0723, 2021-12-13:
[image]

I'm currently running Webcamoid daily build(webcamoid-installer-linux-daily-master-1-x86_64.run)) on garuda linux, kde, x11.  According to the daily build's description, one of the new features of new UI is clicking on the photo preview supposed to open the picture in system's default image viewer but it stopped working after trying 8.8.0 and switching back to daily build.

All I know is it used to work just fine up until I switched back & forth from 8.8.0 to daily build. My default image viewer is LXimage. I changed it to gthumb just to see if it might fix the issue but no luck with this either. I also tried to troubleshoot by executing webcamoid.sh from terminal. This is the error i got after clicking on the photo preview icon:

qt.qpa.plugin: Could not find the Qt platform plugin "xcb" in ""
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.
/usr/bin/xdg-open: line 613: 162175 Aborted (core dumped) kde-open${KDE_SESSION_VERSION} "$1"

Then I found a solution on google from someone who previously had similar issue. His was fixed by installing qtcreator but unfortunately, it didn't work on my machine. yeah so, I'm still stuck with this same issue. Any suggestion would be greatly appreciated. I was forced to switch back to cheese(btw, cheese freezes whenever resolution is set to anything above 640x360. anyone experiencing this issue?) due to this issue. This is the biggest showstopper for me but I really like the new UI of webcamoid and hope to get back to it once this bug is fixed.

Thanks

hipersayanX, 2021-12-21:
Try the latest daily build, it should be fixed.

shawnk0723, 2021-12-22:
Awesome! yeah I just tested it. Everything's fixed and working perfect. Thanks man! :)

shawnk0723, 2021-12-22:
sorry, it was little too quick to close this issue. I tried daily build again this morning but it still won't open the image if I set my default image viewer to gthumb (works with cheese app) for some reason. However, it's working with perfect with lximage now. I'm getting following error when I run from the terminal upon clicking the photo preview icon: 

"kf.service.services: KApplicationTrader: mimeType "x-scheme-handler/file" not found /usr/bin/gthumb: symbol lookup error: /usr/lib/libgtk-3.so.0: undefined symbol: pango_font_family_get_face"

hipersayanX, 2021-12-23:
Hmm...  don't think this bug gonna be fixable at all, the problem is that the LD_LIBRARY_PATH (needed by Webcamoid to work) is conflicting with the externally launched app, maybe setting the rpath on the binaries would allow to set the libraries search path, without requiring LD_LIBRARY_PATH, but I don't hold any hope on this.
Also, just in case, this bug only affects the binary distribution, compiling from source code or installing from the repositories doesn't have this problem.

shawnk0723, 2021-12-23:
Thank you for the quick reply. I tried with source code but still no luck. it's weird that all the other image viewers work except for gthumb. I guess it's my luck. LOL I'll just use other viewer for now. Thanks man. :)

hipersayanX, 2021-12-24:
I will reopen this, because I want to test if setting the rpath works, if not I will close it again.

shawnk0723, 2021-12-24:
> I will reopen this, because I want to test if setting the rpath works, if not I will close it again.

ok, cool! Let me know once it's done. I'll test it out for you. Thanks man! :)

hipersayanX, 2022-01-04:
Can you test it again? setting the rpath seems to have fixed the bug at least for me.

shawnk0723, 2022-01-04:
> 

sure. I'll test it out and let you know soon.

shawnk0723, 2022-01-04:
yes sir!! that was it. that fixed it finally! works perfect with gthumb now. Thanks a lot man! I'm closing this issue now. Happy New Year Bro! :)

closed 2022-01-04 by shawnk0723