webcamoid/webcamoid #475: Poor Documentation - WebCamoid ChromaKey Green Screen Setup

Leemusa, 2021-12-24:
Funny that the first solution your directions suggest is to search for an answer. I've spent hours trying to figure out your software and searching online for a code example and still I can find no explanation / documentation as to how to solve these problems

I downloaded WebCamoid to use it as a virtual camera to add a background to videos.

I am looking for documentation / directions on HOW to set up WebCamoid to use a ChromaKey Green Screen to add a virtual background and create a virtual camera?

I have WebCamoid installed on MacOS Mojave 10.14 but cannot figure out how to do this and I cannot find directions anywhere. In theory I know you need to

1. Add an image for the background
2. Add a filter to process the video and remove the green color from the video
3. Setup WebCamoid as a virtual webcam

Can you please provide a step-by-step for the nube with your software to explain what should be tapped and how. If you provide me with a decent explanation I promise to make a video to explain how WebCamoid si setup.

billeggers182, 2021-12-26:
i also need this to work for me.  I just installed linux mint and used the software manager.  It downloads webcamoid 8.6.1 then tells me there is an upgrade.  I have downloaded the upgrade, but don't understand how to apply the newer version or tell it to upgrade it using the file I downloaded.  I have brain damage/memory challenges from a car crash 5 years ago and probably just can't remember how to do it?  I need to make some videos of me talking about my mental challenges and need to use a green screen product I bought.  Please help me as well if possible people.  I just need a link or advice and I will work on it!

hipersayanX, 2021-12-27:
>I am looking for documentation / directions on HOW to set up WebCamoid to use a ChromaKey Green Screen to add a virtual background and create a virtual camera?

Webcamoid has not such feature yet, and there is an issue already opened for implementing that [feature](https://github.com/webcamoid/webcamoid/issues/250).

@billeggers182

>I have downloaded the upgrade, but don't understand how to apply the newer version or tell it to upgrade it using the file I downloaded

Uninstall the Webcamoid version installed from the repositories, and install it from [the official downloads](https://github.com/webcamoid/webcamoid#downloads), or you can just use the AppImage version which does not need install.

closed 2022-02-15 by hipersayanX